Klipper3d/klipper #4023: UI issue: shouldn't be possible to turn feed rate to 0 with control knob

darkpoetry, 2021-03-07:
Log not included because this is a pretty generic request and I'm just hoping to save a future person a headache and wasted plastic that I just had.

In the LCD tune menu is an adjustment for feed rate. It allows one to kill the print a little too easily with a cheap / course knob selection. Turn it a little too fast and system shutdown.  (S must be above 0.0)

Since there's no conceivable scenario for a user to want to run the print at less than 1% the interface should backstop at 1 and not go to zero and trigger an unrecoverable shutdown.

mcmatrix, 2021-03-08:
Hi @darkpoetry

The current implementation of feed (flow) rate tuning is following.
The maximum (`input_max`) is 200% and minimum (`input_min`) is 0
```ini
[menu __main __tune __flow]
type: input
name: Flow: {'%3d' % (menu.input*100)}%
input: {printer.gcode_move.extrude_factor}
input_min: 0
input_max: 2
input_step: 0.01
realtime: True
gcode:
    M221 S{'%d' % (menu.input*100)}
```
What do you recommend that the new minimum rate should be?

darkpoetry, 2021-03-08:
Ultimately, the user should not be able to select an impossible value. If either flow or speed (feed) rates of zero trigger shutdown, they should not be selectable. 

Given the current granularity of the control (0-100) the minimum selectable should be 1.

holgin, 2021-03-15:
On the other side, I think that 200% is low. I quite often went even to 500% during some tests, for example starting a test print at 60mm/s and finishing at 300mm/s. Maybe this value could be configurable?

mcmatrix, 2021-03-15:
Hi @holgin 

In case you have a specific need for the menu then you can always override the menu item in your printer config.

mcmatrix, 2021-03-21:
Please check and test #4086

closed 2021-05-04 by github-actions[bot]