Klipper3d/klipper #4027: Documentation for raspberry Pi as an MCU and PWM support

gcormier, 2021-03-08:
I'd like to issue a PR to make a note on the Pi documentation page ( https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md ) - that PWM can be accomplished on any GPIO.

I didn't see this mentioned anywhere, so I spent a bit of time yesterday confirming it with a logic analyzer as I designed a Pi hat to control some additional fans. At first I was limiting myself to the hardware PWM pins, but it would appear that doesn't matter with Klipper. Nice to know any pins can be used for PWM - especially as one of the hardware PWM pins is shared with one of the SPI busses.

Let me know and if so I will draft up a PR.

jakep82, 2021-03-08:
https://www.klipper3d.org/FAQ.html#do-i-have-to-wire-my-device-to-a-specific-type-of-micro-controller-pin

gcormier, 2021-03-08:
Well then... damnit :)

closed 2021-03-08 by gcormier