Klipper3d/klipper #4034: Move after pause

TheJoshW, 2021-03-10:
On filament runout, the PAUSE and runout GCODE (PARK) execute correctly, however sometimes after the PARK completes the toolhead moves back to the original X pos. This may be caused by commands being queued from the host after the PAUSE and runout GCODE is queued, this may be corrected by using toolhead.wait_moves() before pausing for pause_delay in filament_switch_sensor.py.
[klippy.log](https://github.com/KevinOConnor/klipper/files/6114713/klippy.log)

schnello, 2021-03-13:
I had the same issues and i solve the problem with: Dont use the Octoprint commands.
I use to start the Pause only "PAUSE"

[gcode_macro M600]
default_parameter_X: 0
default_parameter_Y: 0
default_parameter_Z: 10
gcode:
 SAVE_GCODE_STATE NAME=M600_state
 PAUSE
 #SET_FILAMENT_SENSOR SENSOR=fila ENABLE=0
 SET_IDLE_TIMEOUT TIMEOUT=18000

and for resume:
[gcode_macro RESUME_MACRO]
gcode:
 RESUME
 SET_IDLE_TIMEOUT TIMEOUT=600

KevinOConnor, 2021-03-13:
There is a window where OctoPrint may have a command in transmit when `action:paused` is sent.  That's why `pause_delay` was introduced.  Not sure there's anything that can be done to improve that with OctoPrint as it is a fundamental limitation of g-code.  Using virtual_sdcard (or one of the Klipper specific frontends which use virtual_sdcard) is the only real fix I know of.

-Kevin

closed 2021-04-11 by github-actions[bot]