Klipper3d/klipper #4038: PID calibration bug

SilverWx, 2021-03-11:
Hello, I tried to calibrate my PID but the software does not wait multiple sinusoid before ending the process. It goes to the temp and once the temp is reached it stop PID calculation and give no functionnal value.
[klippy (7).log](https://github.com/KevinOConnor/klipper/files/6122446/klippy.7.log)

jakep82, 2021-03-11:
You have a hardware issue. The temperature on your hotend is fluctuating rapidly even when heat is turned off.

SilverWx, 2021-03-11:
Okay, thank you. I have checked multiple time wiring before and was desperate until I tried to turn off my led and the temperature started to stabilize. I have to change the frequency of my led, it explained why I naver had this issue on marlin

closed 2021-03-14 by github-actions[bot]