Klipper3d/klipper #4041: Internal error during connect: MCU_TMC_SPI instance has no attribute 'name'

case-tm77, 2021-03-11:
Hello all,

after i updated klipper an hour ago, i have this error showing up.
"Internal error during connect: MCU_TMC_SPI instance has no attribute 'name' "

i made little changes to my config (new rotational_distance for the new installed extruder) so I'm sure that the changes have nothing to do with the error.

and the log is pointing to a py script.

i checked the config, but found no error, may by its in the script?

[klippy.log](https://github.com/KevinOConnor/klipper/files/6125618/klippy.log)

mouring, 2021-03-11:
FYI, I'm seeing the same thing.  SKR v1.3 using TMC2130s.  (Was a working configuration) The error is:

Full logs posted, but this is the python stack trace:

Configured MCU 'mcu' (1024 moves)
Starting heater checks for heater_bed
Unhandled exception during connect
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 151, in _connect
    cb()
  File "/home/pi/klipper/klippy/extras/tmc.py", line 267, in _handle_connect
    self._init_registers()
  File "/home/pi/klipper/klippy/extras/tmc.py", line 191, in _init_registers
    self.mcu_tmc.set_register(reg_name, val, print_time)
  File "/home/pi/klipper/klippy/extras/tmc2130.py", line 243, in set_register
    "Unable to write tmc spi '%s' register %s" % (self.name, reg_name))
AttributeError: MCU_TMC_SPI instance has no attribute 'name'

[klippy.log](https://github.com/KevinOConnor/klipper/files/6126164/klippy.log)

mouring, 2021-03-11:
FYI if I do: "git reset --hard 3f07b16b" in the ~/klipper directory I'm back to working order.  So I suspect something in the "tmc2130: Verify SPI register writes" commit is causing this error to be thrown.

KevinOConnor, 2021-03-11:
Thanks.  I fixed the error message (commit bf5c2505).  If you now start to get "unable to write tmc spi register" errors then see: https://www.klipper3d.org/TMC_Drivers.html#i-keep-getting-unable-to-write-tmc-spi-stepper_x-register--errors

-Kevin

mouring, 2021-03-11:
Can confirm that fixed it for me. I was just trying to figure out what to set self.name to. =)

Thanks

case-tm77, 2021-03-12:
It worked also for me.

Thank you!

closed 2021-03-12 by case-tm77