Klipper3d/klipper #4042: Move menu.cfg to the configuration directory

NameOfTheDragon, 2021-03-11:
The file `klippy/extras/display/menu.cfg` may need to be changed to adapt to local requirements.
For example, I like to modify the filament load/unload entries as I have a customized printer and the stock configuration doesn't work well.
However, this fil is inside the git repository so editing it marks the repository as `dirty` in Fluidd and then it can't be updated from the UI.

Could this file not be moved into the configuration directory, alongside `printer.cfg` so that it is not considered part of the source code?

jakep82, 2021-03-11:
You should never edit menu.cfg.  All changes you make to your menu should be contained in your printer.cfg file so they are persistent across updates.

NameOfTheDragon, 2021-03-12:
I am pleased to learn that. It seems wrong editing something that's part of the source code, so I'm happy to learn that it can be done in `printer.cfg`. However, it is not obvious from any of the documentation how to do that. I have questions! Do I just copy the relevant sections from `menu.cfg` and paste them into `printer.cfg`? Will these then just override whatever is in `menu.cfg`? How do I _remove_ a menu entry (such as the OctoPrint menu, which I don't use)?

theopensourcerer, 2021-03-12:
@NameOfTheDragon 

> A default set of menus are automatically created. One can replace or extend the menu by overriding the defaults in the main printer.cfg config file.

https://www.klipper3d.org/Config_Reference.html#menu

jakep82, 2021-03-12:
You can also read the comments at the top of the menu.cfg file which explains this.

NameOfTheDragon, 2021-03-12:
> You can also read the comments at the top of the menu.cfg file which explains this, but this isn't the place for support questions. Please use the Discord for that.

Ah, there you go, that comment was added after I made my copy of the config file, so it wasn't in mine. All good, I have it working now. Thank you for your responses.

closed 2021-03-12 by NameOfTheDragon