Klipper3d/klipper #4044: M112 doesn't stop motion during homing

jstkatz, 2021-03-12:
I'm trying to setup sensorless homing on a Sapphire Plus that I've switched over to a SKR V1.4 Turbo with TCM2209 drivers. When I do a G28 x and the machine keeps ramming the wall, I can enter a M112 but so far each time the homing timeout occurs before the M112 works, this leads to ~7 seconds of wall ramming. 

I'm entering both commands through the console in fluidd. 

I'm aware some gcode implementations execute strictly in order so this could be a lack of feature rather than a bug, however I wouldn't think so based on the language in the sensorless homing portion of [https://github.com/KevinOConnor/klipper/blob/master/docs/TMC_Drivers.md](url)

[klippy (1).log](https://github.com/KevinOConnor/klipper/files/6127367/klippy.1.log)

Arksine, 2021-03-12:
M112 won't jump the gcode queue when entered from Fluidd.  You must click on the emergency stop icon.

jstkatz, 2021-03-12:
Ok, if Fluidd doesn't even send the M112 until it sees the G28 finish then this is really a Fluidd/User issue, I hadn't actually recognized what the e-stop button was

closed 2021-03-12 by jstkatz