Klipper3d/klipper #4046: TMC2130 spi bus issue since v0.9.1-3xx

jperrin72, 2021-03-12:
Since I upgraded to the latest branch version (V0.9.1-306) my printer refuses to home and crash.

I have a custom 3D printer based on a TEVO tarentula clone, running FYSETC F6 board with 6 TMC_2130 drivers and 5 steppers connected (x,y,z1,z2,extruder) with sensorless homing. 
Looking at the log it seems like the SPI bus can't communicate with the TMC2130 driver anymore.

*** Doing a rollback to a former klipper version solves the issue. ***

Here are the obvious evidence I can find into klippy.log:

Starting heater checks for extruder
TMC stepper_x failed to init: Unable to write tmc spi 'stepper_x' register CHOPCONF
TMC stepper_y failed to init: Unable to write tmc spi 'stepper_y' register CHOPCONF
TMC stepper_z failed to init: Unable to write tmc spi 'stepper_z' register CHOPCONF
TMC stepper_z1 failed to init: Unable to write tmc spi 'stepper_z1' register CHOPCONF
TMC extruder failed to init: Unable to write tmc spi 'extruder' register CHOPCONF
...
Stats 5360.2: gcodein=0 mcu: mcu_awake=0.016 mcu_task_avg=0.000126 mcu_task_stddev=0.000084 bytes_write=4893 bytes_read=8893 bytes_retransmit=0 bytes_invalid=0 send_seq=223 receive_seq=220 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=550 stalled_bytes=0
freq=15983860  mcu: temp=30.0 heater_bed: target=0 temp=15.4 pwm=0.000 chamber_heater: target=0 temp=15.7 pwm=0.000 print_time=188.073 buffer_time=1.394 print_stall=0 extruder: target=0 temp=16.5 pwm=0.000 sysload=0.39 cputime=5.909 memavail=846136
Unable to write tmc spi 'stepper_x' register GCONF
Transition to shutdown state: Unable to write tmc spi 'stepper_z' register CHOPCONF
Dumping gcode input 0 blocks
gcode state: absolute_coord=True absolute_extrude=True base_position=[0.0, 0.0, 0.0, 0.0] last_position=[315.5, 0.0, 9.98, 0.0] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=1.0 speed=25.0
Reactor garbage collection: (5360.266133684, 0.0, 0.0)
MCU 'mcu' shutdown: Command request
clocksync state: mcu_freq=16000000 last_clock=2981446935 clock_est=(5344.717 2738692506 15983860.382) min_half_rtt=0.001294 min_rtt_time=5343.166 time_avg=5344.716(62.301) clock_avg=2738692506.663(995810878.700) pred_variance=89761140.149
Dumping serial stats: bytes_write=5754 bytes_read=9219 bytes_retransmit=0 bytes_invalid=0 send_seq=242 receive_seq=242 retransmit_seq=0 srtt=0.004 rttvar=0.001 rto=0.025 ready_bytes=0 stalled_bytes=16
Dumping send queue 100 messages
Sent 0 5339.161731 5339.156652 51: seq: 1e, st7920_send_data oid=0 data='\x00\x00\x00\x00\x0f\xe0\x1f\xf8\x1f\xfc?\xfc<<<<', st7920_send_cmds oid=0 cmds='H', st7920_send_data oid=0 data='<<<<?\xfc?\xf8\x1f\xf8\x07\xf0\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='P'
Sent 1 5339.168013 5339.165973 51: seq: 1f, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00R\x94\x00\x00R\x94\x00\x00R\x94', st7920_send_cmds oid=0 cmds='X', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_
cmds oid=0 cmds='`'
Sent 2 5339.170053 5339.165973 51: seq: 10, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='h', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7
920_send_cmds oid=0 cmds='p'
Sent 3 5339.172173 5339.165973 53: seq: 11, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='x', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7
920_send_cmds oid=0 cmds='&\x80\x80'
Sent 4 5339.178303 5339.176183 53: seq: 12, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='\x80\x88', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0', st7920_send_cmds oid=0 cmds='\x81\x80'
Sent 5 5339.180423 5339.176183 53: seq: 13, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='\x81\x88', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0', st7920_send_cmds oid=0 cmds='\x82\x80'
Sent 6 5339.182543 5339.176183 53: seq: 14, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='\x82\x88', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0', st7920_send_cmds oid=0 cmds='\x83\x80'
Sent 7 5339.184663 5339.182464 53: seq: 15, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='\x83\x88', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0', st7920_send_cmds oid=0 cmds='\x84\x80'
Sent 8 5339.190942 5339.188822 53: seq: 16, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='\x84\x88', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0', st7920_send_cmds oid=0 cmds='\x85\x80'
Sent 9 5339.193062 5339.188822 53: seq: 17, st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00', st7920_send_cmds oid=0 cmds='\x85\x88', st7920_send_data oid=0 data='\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x0
0', st7920_send_cmds oid=0 cmds='\x86\x80'

jperrin72, 2021-03-12:
As an additional test, I put the ouput of a DUMP_TMC STEPPER=stepper_x after a fresh start and before homing :

DUMP_TMC stepper_x
========== Write-only registers ==========
IHOLD_IRUN: 00081010 IHOLD=16 IRUN=16 IHOLDDELAY=8
TPOWERDOWN: 00000000
PWMCONF:    000602f0 PWM_AMPL=240 PWM_GRAD=2 pwm_freq=2 pwm_autoscale=1
COOLCONF:   000a0000 sgt=10
========== Queried registers ==========
GCONF:      ffffffff I_scale_analog=1(ExtVREF) internal_Rsense=1 en_pwm_mode=1 enc_commutation=1 shaft=1(Reverse) diag0_error=1 diag0_otpw=1 diag0_stall=1 diag1_stall=1 diag1_index=1 diag1_onstate=1 diag1_steps_skipped=1 diag0_int_pushpull=1 diag1_pushpull=1 small_hysteresis=1 stop_enable=1 direct_mode=1 test_mode=1
GSTAT:      ffffffff reset=1(Reset) drv_err=1(ErrorShutdown!) uv_cp=1(Undervoltage!)
IOIN:       ffffffff STEP=1 DIR=1 DCEN_CFG4=1 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0xff
TSTEP:      ffffffff TSTEP=1048575
XDIRECT:    ffffffff
MSCNT:      ffffffff MSCNT=1023
MSCURACT:   ffffffff CUR_A=-1 CUR_B=-1
CHOPCONF:   ffffffff toff=15 hstrt=7 hend=15 fd3=1 disfdcc=1 rndtf=1 chm=1 TBL=3 vsense=1 vhighfs=1 vhighchm=1 sync=15 MRES=15(0usteps) intpol=1 dedge=1 diss2g=1
DRV_STATUS: ffffffff SG_RESULT=1023 fsactive=1 CS_ACTUAL=31 stallGuard=1 ot=1(OvertempError!) otpw=1(OvertempWarning!) s2ga=1(ShortToGND_A!) s2gb=1(ShortToGND_B!) ola=1(OpenLoad_A!) olb=1(OpenLoad_B!) stst=1
PWM_SCALE:  ffffffff PWM_SCALE=255
LOST_STEPS: ffffffff LOST_STEPS=1048575

jperrin72, 2021-03-12:
after a few tries and rollback, it seems like the regression was introduced around the 27/02 - see below

git checkout 11b9b72 #KO v0.9.1-306-g11b9b72b 11/03
git checkout bc2096f #KO v0.9.1-290-gbc2096f5 02/03
git checkout 4d3d25b #KO v0.9.1-289-g4d3d25b1 01/03
git checkout 4d3d25b #KO v0.9.1-289-g4d3d25b1 01/03
git checkout ddb795b # KO v0.9.1-282-gddb795bc 28/02
git checkout 86c9434 # KO v0.9.1-276-g86c94346 27/02
git checkout 3fc72da # OK 0.9.1-264-g3fc72da9 27/02
git checkout c7688c6 # OK v0.9.1-262-gc7688c6b 26/02
git checkout a8742e9 # OK v0.9.1-261-ga8742e98 23/03

KevinOConnor, 2021-03-12:
The Fysetc F6 board has a device which blocks communication on the SPI bus unless it is disabled.  Add the following to your config: https://github.com/KevinOConnor/klipper/blob/106d1d2a2aea4d8b15cb22d7e4382d417fd3c554/config/generic-fysetc-f6.cfg#L95

A high-level desription of the issue is at: https://www.klipper3d.org/TMC_Drivers.html#i-keep-getting-unable-to-write-tmc-spi-stepper_x-register--errors

-Kevin

jperrin72, 2021-03-12:
That indeed did the trick! Thanks for your great support Kevin :)

closed 2021-03-12 by jperrin72

cAlbiez, 2021-03-14:
Hi, 

I'm having the same issue but with a Ramps 1.6 plus from bigtreetech with an original Arduino Mega underneath.
4x TMC 2130 all configured and working prior to the update.
no LCD or SD-Card attached.

tried the above Fysetc F6 fix just to try (as that is also 2560 based), removed the whole EXP1 and EXP2 section from the config,
still as soon as i upgrade from v0.9.1-179 to v0.9.1-317 i get the "tmc spi 'stepper_x' register GCONF" error.

best,

Christoph

KevinOConnor, 2021-03-15:
Best advice I have is at https://www.klipper3d.org/TMC_Drivers.html#i-keep-getting-unable-to-write-tmc-spi-stepper_x-register--errors .  If you're getting that error it indicates the SPI wiring or configuration is not correct.  Double check your wiring - in particular the wiring of the MISO pin.

-Kevin

cAlbiez, 2021-03-15:
Hi,

thanks for coming back, I checked there are no additional devices, the SPI bus is routed through the RAMPS 1.6+ PCB so no flying wires, no SD or LCD attached.
What did change in software? The hardware is working on version 0.9.1-179.
Did the SPI Speed or anything else change?

I tried setting ar53(PB0) as static as described above as this is SS for arduino mega 2560 but this does not change anything.
I also tried setting:

spi_speed: 400000 // tried 8000000, 4000000, 2000000, 1000000
spi_bus: spi //tried with and without this line no change

all with the same result:
_!! Unable to write tmc spi 'stepper_x' register GCONF

Unable to write tmc spi 'stepper_z' register CHOPCONF
Once the underlying issue is corrected, use the
"FIRMWARE_RESTART" command to reset the firmware, reload the
config, and restart the host software.
Printer is shutdown_

I attach the klippy.log's last entry 
[klippy.txt](https://github.com/KevinOConnor/klipper/files/6143846/klippy.txt)

full log including a longer print, done on 0.9.1-179 at the beginning, the last few entries are after the upgrade to
[klippy.zip](https://github.com/KevinOConnor/klipper/files/6143891/klippy.zip)

I will switch back to  5d960aef from 2021-01-24 but would really like to not be stuck, if I can help please let me know.

Please help,
best regards,
Christoph

KevinOConnor, 2021-03-15:
There was a recent change to the Klipper TMC2130 code to confirm that the code is actually able to communicate with the driver.  This is not related to spi speed.  Unfortunately, your drivers are not able to perform two way communication with the micro-controller.

If your board is the pcb described at: https://github.com/bigtreetech/ramps-1.6/blob/master/Ramps1.6/hardware/R6%2B-SCH.pdf , then make sure you have the "MOSI" jumpers in place.  You'll also need to confirm that your stepsticks have the MOSI pin routed to the appropriate stepstick pin (pin 5 or 6) and that neither of the pins routed to MOSI (pin 5 and 6) are being grounded by the stepstick.

-Kevin

cAlbiez, 2021-03-15:
Thanks,

and sorry :( 

I went and checked, all jumpers were in place (only X,Y,Z and E0 are in use)
[image]

And all jumpers have a connection to the corresponding pins on the drivers.

measuring pin 5&6 against GND they show ~180kOhms when unplugged, when all stepsticks are pluggedin i measure ~50kOhms against gnd.

[image]

not sure if this could be releated but to flash the arduino i had to comment out the i2c in the .config
`#CONFIG_HAVE_GPIO_I2C=y`

or i get:

```
src/i2ccmds.c:14:23: error: field ‘i2c_config’ has incomplete type
     struct i2c_config i2c_config;
                       ^
src/i2ccmds.c: In function ‘command_config_i2c’:
src/i2ccmds.c:23:23: warning: implicit declaration of function ‘i2c_setup’ [-Wimplicit-function-declaration]
     i2c->i2c_config = i2c_setup(args[1], args[2], addr);
                       ^
src/i2ccmds.c: In function ‘command_i2c_write’:
src/i2ccmds.c:35:5: warning: implicit declaration of function ‘i2c_write’ [-Wimplicit-function-declaration]
     i2c_write(i2c->i2c_config, data_len, data);
     ^
src/i2ccmds.c: In function ‘command_i2c_read’:
src/i2ccmds.c:48:5: warning: implicit declaration of function ‘i2c_read’ [-Wimplicit-function-declaration]
     i2c_read(i2c->i2c_config, reg_len, reg, data_len, data);
     ^
make: *** [Makefile:64: out/src/i2ccmds.o] Error 1
```

Thanks for your help.

Kind regards,

Christoph

KevinOConnor, 2021-03-15:
What stepstick boards are you using?

-Kevin

cAlbiez, 2021-03-15:
watterot tmc2130

KevinOConnor, 2021-03-15:
Looks to me like the board has a conflict with the DCO and MISO pins.  That is, the RAMPS v1.6+ board is connecting MISO to both MISO and DCO of the tmc2130.  I suspect that conflict results in the micro-controller not being able to succesfully communicate.  

https://learn.watterott.com/silentstepstick/pinconfig/tmc2130/

Note the text: `The DCO pin is an output, keep attention that it is not connected to SDO, because some mainboards bridge these pins (also called SLP + RST).` on the above page.

It's just a guess, but if you wish to pursue that, you could cut the DCO pin (or otherwise remove it) from all the stepsticks.  It would be necessary to remove it from all of the stepsticks, but you could try with only one altered stepstick inserted into the board to verify it fixes the issue before altering the other sticks.

Sorry - I know it's a pain - but you really do want 2-way communication with these stepsticks.  Only having 1-way communication can cause all sorts of "strange" errors.

-Kevin

cAlbiez, 2021-03-15:
OMG! !!!!! that could be it!
[image]

did you have your fair share of praise today? if not be praised!

I will report, 

you are amazing!

Chris... (in a hurry to the soldering iron)

cAlbiez, 2021-03-15:
Can confirm this was the issue, I would never have found it, thanks a lot.

Best Regards, Christoph

BastR, 2021-08-07:
@cAlbiez I have the same problem, I will try to cut the DCO pin. Is it possible that you share with me your printer.cfg file to start on a good basis?

Thank you very much