Klipper3d/klipper #4049: [support] z_endstop offset measure with detachable probe

robvoi, 2021-03-13:
Hey. I hope someone can help me out here.
I use a detachable probe for tilt calibration and bed mesh (setting bed center to 0).
Is there way to set the z_endstop offset with it? The distance z_endstop to probe is fixed. So it should be possible to calculate the current distance between bed and probe.
What is a good way to do this?

robvoi, 2021-03-13:
Dear Bot.
I fully get that Discord should be used. Didn't get an answer there though. That's why I tried here.
Very good software - so thanks in any case. :-)

KevinOConnor, 2021-03-19:
Alas, I don't know the answers to your questions - you'll likely need to implement it and tell us about it.

-Kevin

closed 2021-03-19 by KevinOConnor