Klipper3d/klipper #4054: Temperature sensor "NTC 100K beta 3950" is very inaccurate

ssotangkur, 2021-03-15:
The built in `sensor_type`  NTC 100K beta 3950 is too inaccurate to be used and shouldn't be provided unless fixed. The algorithm used, assumes the beta is a constant 3950 throughout the temp range, but in reality none of these types of thermistors spec their beta in the critical temp ranges needed for 3d printing. Instead, a linear interpolation of the resistance/temperature chart with several datapoints should be used. [Here](https://www.makeralot.com/download/Reprap-Hotend-Thermistor-NTC-3950-100K.pdf) is such a datasheet. 

Using a thermocouple equipped multi-meter I first confirmed that it matched the temp reading of the printer using its original `EPCOS 100K B57560G104F` and confirmed they were within 1C of each other over multiple temps throughout their range. To do so I placed the thermocouple into a slot next to the heater cartridge what was just wide enough to fully embed the probe inside. I then replaced the thermistor with an HT-NTC100K Beta 3950 sensor from Amazon. I repeated the temp check but this time noticed a very wide discrepancy (15-20C over-reporting). I tried various calibration methods but found that using the table provided to create a custom `[adc_temperature]` with the same resistance, temperature values, produced a highly accurate sensor. This brought the calculation back to within 1C of the thermocouple readings.

## The Ask
- remove line:  https://github.com/KevinOConnor/klipper/blob/3233ec08f4a7165d54413e7a24b0021053604053/klippy/extras/thermistor.py#L118
- Add it back as a "resistance sensor" someplace like here: https://github.com/KevinOConnor/klipper/blob/3233ec08f4a7165d54413e7a24b0021053604053/klippy/extras/adc_temperature.py#L281

KevinOConnor, 2021-03-15:
I'm not sure what you're reporting.  It is true that using beta isn't as accurate as steinhart-hart or a custom temperature table, but some sensors are defined as "beta 3950" - that is, the manufacturer produced the sensor to fit that curve as best as they can and the only temperature information they provide is that curve.  The "makeralot" link is just one thermistor - it isn't all "beta 3950" thermistors.

So, although "beta 3950" may not be very "accurate" it is more accurate than anything else we have.

-Kevin

ssotangkur, 2021-03-16:
What if literally every NTC 100k beta 3950 you can buy corresponds to the same datasheet? 

Here are the ones I could find:

https://hc-maschinentechnik.de/bilder/Datenblatt/Thermistor-Datenblatt.pdf (same values)
https://forum.duet3d.com/assets/uploads/files/1541668164665-ntc-thermistor-r-t-table.pdf (same, but looks like a copy of the Reprap one)
https://www.tme.eu/Document/f9d2f5e38227fc1c7d979e546ff51768/NTCM-100K-B3950.pdf (125C Max not suitable for 3d printing)
https://www.tdk-electronics.tdk.com/inf/50/db/ntc/NTC_Probe_ass_K45.pdf (125C Max not suitable)

Every forum post about using NTC 100k 3950 thermistor comments on how the Beta is only valid for low temps and becomes very inaccurate at normal printing temps. This is a characteristic of both the thermistor and how the industry chooses to characterize them. An "NTC 100k beta 3950" is an NTC 100k beta 3950 with a beta of 3950 at low temps and a different beta at other temps. 

If you choose to mirror an NTC 100k beta 3950 datasheet, it's still an NTC 100k beta 3950 by all those concerned. And for those who aren't concerned, it will just work as they intended (isn't that what we all want?).

In the places where the ideal beta 3950 diverges from the datasheet, the datasheet will be close in all cases.

KevinOConnor, 2021-03-19:
Interesting.  We can certainly add another default thermistor if you know of an appropriate name.  It isn't necessary to use adc_temperature, as steinhart-hart provides excellent results with just three temperature/resistance measurements.  FYI, it is possible to compare sensors using the scripts/graph_temp_sensor.py tool (some info on using these graphing tools is in docs/Debugging.md ).

-Kevin

ssotangkur, 2021-03-25:
I'll have to give those tools a try. Meanwhile I think `Reprap Thermistor NTC 3950 100K` is as good of a name as any.

eddietheengineer, 2021-05-02:
Hello @ssotangkur, thank you for opening this issue and starting the discussion!

@KevinOConnor, I've had similar concerns with the beta3950 as you have--I wasn't sure how consistent they were vs. each other, but with the 3 references to the same nominal resistance that were provided, I think we can be fairly confident that the data sheets are more accurate than using the beta approximation. 

I've gone ahead and plotted the difference in nominal (and tolerances) of this thermistor using the given datasheet(s) and compared to the current Klipper implementation using the beta approximation:

[image]

At 250C, we see about a 15C error just from the beta approximation--which happens to be more substantial than the entire thermistor tolerance. For a chamber thermistor or bed thermistor, the error isn't as pronounced, but I've seen more and more people using this thermistor for their hotend as well--and for good reason. The thermistor tolerances are substantially better than the other common hotend thermistor (104GT/NT), as long as the datasheet/calculation is correct.

I'm not sure what the "best" approach here is. The least intrusive would be to add a new thermistor definition such as the "Reprap Thermistor NTC 3950 100K", though I can see the argument to correct the currently used thermistor definition. There is precedence here as well with the calculation correction for the PT100/PT1000 RTD in Klipper last year.

The proposed update would be:
temperature1: 20
resistance1: 125245
temperature2: 80
resistance2: 12540
temperature3: 220
resistance3: 396

I specifically picked these values to minimize the 3 point Steinhart-Hart error vs. the datasheet around the three most critical areas--room temperature (20C), print bed temperature (60-100C), and hotend temperature (190-250C). Below is the resistance error, column 2 is the proposed improvement, column 3 is the current beta estimation. 

[image]

KevinOConnor, 2021-05-03:
Thanks.  The resulting sensor definition would be very close to the existing "EPCOS 100K B57560G104F" sensor.  So, is it worth adding a new definition instead of just pointing users to an existing definition?

See graph below (with "RepRap 3950" corresponding to Eddie's table above):

`./scripts/graph_temp_sensor.py -s 'EPCOS 100K B57560G104F,ATC Semitec 104GT-2,NTC 100K beta 3950,RepRap 3950'`

[image]

-Kevin

ssotangkur, 2021-05-03:
I think the axes of the graphs need to be inverted to see how much of an error there is. Even with the 3 point Steinhart-Hart we get some error. Why not use the datasheet values verbatim (all 300+ of them) and eliminate any error?

Sineos, 2021-05-04:
> Why not use the datasheet values verbatim (all 300+ of them) and eliminate any error?

I do not think that it would add to a real-world precision. All non-calibrated probes will have an error. Even a PT100 Class B will have an error of +/- 1.3 °C at 200 °C according to EN 60751. For NTC no standards are even existing

Additionally you typically have:
- Unknown, non calibrated measuring circuits
- Unknown cable length
- Ageing / degradation effects (especially NTC)

With our consumer grade equipment, precise temperature measurement is an illusion IMO.
Of course, I agree that the "approximation" should be as good as possible.

Side note:
If someone knows a source for a 4-wire PT100 Class A that fits a typical hotend, I would love to hear about it.

ssotangkur, 2021-05-05:
In my makeshift testing I found the difference between the NTC using 300 pts and my thermocouple on my multi-meter to never diverge more than 1C. Of course that doesn't mean that they must both be accurate, but chances are that they are since thermocouples and thermistors need to be calibrated differently and the places where they drift are not likely to be in the same spots with the same magnitude of error. 

So I would say (despite only having a sample size of 1) that there are realizable precision gains to be had.

Also considering that it's much harder for people to enter 300 pts into their klipper config, I don't see why we wouldn't opt for this approach that yields the best approximation. A built-in definition is less valuable if it is less accurate and only saves you 3 lines in the file. But if a built-in was as accurate as it could be and it saves you 300+ lines, then it becomes really valuable as a built-in.

KevinOConnor, 2021-05-06:
I guess the main question for me is - if I buy some random "beta 3950" thermistors, how do I know they follow the temperature curve specified at https://forum.duet3d.com/assets/uploads/files/1541668164665-ntc-thermistor-r-t-table.pdf as opposed to some other curve?

For example, I recently bought a ten pack of thermistors from amazon ( https://www.amazon.com/gp/product/B07V6YBFSY ).  How can I know what to declare these as (besides measuring them at high temperatures with a known good reference)?

-Kevin

ssotangkur, 2021-05-07:
I'd assume if you had an existing thermister that was know to be accurate (like maybe you trust the stock one that came w/ your printer) then see how low you can set the hotend till the extruder starts to skip. Then when using the unknown thermister, try repeating the experiment and see if the extruder skipping temp is at the same point.

Ultimately, there is no way to measure your thermister without some known reference. If you really want to get technical, one could try melting tin in your oven (239C) and when the tin melts into a pool around the probe but not completely melted, measure the resistance.

KevinOConnor, 2021-05-27:
Okay, FWIW I'm no longer sure what's being recommended here.  I think it may help to put together a pull request with the desired change.

Cheers,
-Kevin

DaStivi, 2021-06-20:
Is there any update, or fix for these thermistor type? Can use another more precise definition?!

Printing the same filament on one printer with 235 and with the trianglelabs B3950 I am at 260 now... Not done any measuring (yet) but that's definitely not correct

ReXT3D, 2021-06-27:
To add some additional information and perhaps help @DaStivi directly, I profiled the Trianglelab NTC100K B3950 against a PT100 connected via the Duet MAX31865 board.  Both sensors were installed in the same heat block and measured at several temperatures before I fit a curve and generated the following data for the Triangle Lab thermistor:

`[thermistor Trianglelab NTC100K B3950]
## values calibrated against a PT100 reference
temperature1: 25.0
resistance1: 103180.0
temperature2: 150.0
resistance2: 1366.2
temperature3: 250.0
resistance3: 168.6`

Without the above I was getting up to 15 deg.C temperature errors (vs. PT100 reference) at PETG printing temperatures.

KevinOConnor, 2021-12-01:
FYI, this should now be addressed with the merge of #4859 .

-Kevin

closed 2021-12-08 by github-actions[bot]