Klipper3d/klipper #4055: Scheduled digital out event will exceed max_duration

tarsonis123, 2021-03-15:
Out of nowhere I got this error and the connection was closed leaving the hot end without cooling... Restart command didn't lead to a new connection to the printer. 
I run Klipper on a SKR E3 2.0 via GPIO since some week with more or less ok prints. Never had this message. Coincidence or not I switched to a 0.8 nozzle with Klipper the first time. Changed the cfg nozzle size before. I don't know if this has something to do with it.
I had to shut down the printer while a print started today because the emergency button/command wasn't reacting.  Since then this error showed up in the Octo logs.

How can this be resolved?

[octoprint.log](https://github.com/KevinOConnor/klipper/files/6142187/octoprint.log)
[klippy.log](https://github.com/KevinOConnor/klipper/files/6142164/klippy.log)

KevinOConnor, 2021-03-15:
Sorry, but we don't use github to diagnose problems on individual printers.  That said, your logs indicate a large number of communication errors - that generally indicates wiring or voltage problems.

-Kevin

tarsonis123, 2021-03-15:
Don't feel sorry. I wonder why this human got help for the same issue then
https://github.com/KevinOConnor/klipper/issues/3635

The behavior in communication with Klipper and the boards firmware was an error. Not obviously an individual one or user error.

jakep82, 2021-03-15:
The contact document was updated after the issue you linked (which is for a different controller board with a known manufacturer firmware defect) specifically to eliminate the large number of individual support requests.

https://github.com/KevinOConnor/klipper/pull/3762

closed 2021-03-19 by KevinOConnor

B0rax, 2021-11-02:
I have this exact same issue with an raspberry pi zero (and a zero 2) connected via GPIO to an SKR Mini.
I tried rerouting the cables, updating everything multiple times, changing the power supply of the SKR and so on...

Did you find a solution for your problem @tarsonis123 ?