Klipper3d/klipper #4058: printer.probe.last_z_result returns 2nd last result

Nikongen, 2021-03-16:
Hi there,

I have a problem with the _printer.probe.last_z_result_ command. In my setup its not returning the last, but the 2nd last probe result. (see screenshot, the last echo command prints the value of printer.probe.last_z_result). 
Before the second call of the _probe_z_ macro, I turned the bed screws quite a bit, so that one could tell 2nd from 1st probe result. 

I've checked the klipper/extras/probe.py, but cloudn't find any obvious reason for this behavior...

Here is the config snippet i use for probing: (note this is a lab machine for some testing)
```
[probe]
pin: ^!ar23
samples: 1
samples_tolerance: 1
sample_retract_dist: 10
x_offset: 0.0
y_offset: 0.0
z_offset: 0.0

[gcode_macro probe_z]
gcode:
	M109 S210 # probing with electrical contact nozzle to sensing plate
	SET_GCODE_OFFSET Z=0
	level_point POINT=0 DISTANCE=20 # moves to xy of 1st levelpoint, lowers to z=20
	PROBE
	G1 Z30
	RESPOND TYPE=echo MSG="{"Z1 Offset: %.3f" % (printer.probe.last_z_result)}"
```
[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/6148078/klippy.log)

FHeilmann, 2021-03-16:
The problem is that the entire macro g-code gets jinja evaluated at the time the macro is called. This means `(printer.probe.last_z_result)` gets substituted with the probe value when you type `probe_z` into the terminal, not when that line of gcode is reached.

Nikongen, 2021-03-17:
That's just what i was looking for. 
Thanks for Your help!

closed 2021-03-17 by Nikongen