Klipper3d/klipper #4060: i keep getting MCU 'mcu' shutdown: Rescheduled timer in the past

PabloAVazquez, 2021-03-16:
After successfully calibrating resonances printing the tower at 100mm/s and pressure advance at that speed too, i went on to print again a resonance tower with pressure advance enabled at 0.7, speeds of 150mm/s and accelerations of 4000 and it printed fine on an atmega2560 based trigorilla 1.1 board (TMC2209 on UART on all axis, X, Y, dual Z, E all with 16 microsteps), I went on then to print some parts... after a couple of successful prints, got an MCU shutdown so I reduced speeds to 100 and acceleration to 2000, reduced microsteps on both Z and extruder to 8, after printing fine some parts it came up again with MCU shutdown, so I reduced print speed to 80 and accel to 1500, some parts printed fine but then again, MCU shutdown, reduced accel to 1000, got rid of wipe on retract, and yet again I get MCU shutdown. Some parts that fail are even repeats of a successfully printed part without touching anything at all!

I'm attaching the logfile.

Thank you very much for your help!

Pablo

[klippy.zip](https://github.com/KevinOConnor/klipper/files/6152498/klippy.zip)

KevinOConnor, 2021-03-16:
You should update to the very latest version of the code - there was a recent fix in this area (commit 9572ad43).

-Kevin

PabloAVazquez, 2021-03-17:
ah perfect!!! installed it and during a 5 hour print with speeds set at 100mm/s, acceleration set to 4000 and pressure advance enabled, the CPU load averaged 42 with spikes of 45%. Thank you very much Kevin!!!

closed 2021-03-19 by KevinOConnor