Klipper3d/klipper #4063: "Timer too close"-Error either on startup or during Z-Leveling

PaulRitter, 2021-03-17:
Printer: Anycubic Mega Zero 2.0 (From what i can gather essentially a CR10/Ender 3 clone, i am using a copy of the cr10 config)
Klipper & Octopi running on a Raspberry Pi 3+ 512MB
The cable connecting them is roughly 2 meters of length, so that might be a cause (?)
I have a BTT installed & am using it for Z-Leveling

I get a "Timer too close" Error somewhat inconsistently on:
- Startup
- During Z-Leveling when the toolhead moves downwards
  Sidenode: this *tends* to happen more consistently the longer the Move is, altough i just had it fail again after moving ~2mm
  I did find error reports about it happening on longer Z moves, but I don't think thats whats happening here
  
[klippy.log](https://pastebin.com/JSyHjrcw)

Thanks in advance for the help and creating this software. I am really happy with its performance & expandability so far, I hope to get around contributing someday! :)

KevinOConnor, 2021-03-19:
Sorry, but we don't use github to help diagnose individual printer problems.  That said, something on your host computer is using excessive CPU usage - you'll need to track that down and fix it.

-Kevin

closed 2021-03-19 by PaulRitter

MattByczko, 2021-09-11:
@PaulRitter Did you ever sort this out?

PaulRitter, 2021-09-12:
@MattByczko my specific problem was simply because my raspberry pi 1b did not have enough cpu power to handle it. I disabled camd (`sudo systemctl disable camd`) and never had any more issues since.