Klipper3d/klipper #4064: Timer too close in the middle of the print

Yury-MonZon, 2021-03-17:
This gcode file just printed fine, I've run it again and in the middle of the print it stopped. Then I hit restart firmware and only after that I realized that I should save the log.
Log included.
Please help.
Thanks a lot!

[klippy.log.zip](https://github.com/KevinOConnor/klipper/files/6159254/klippy.log.zip)

I've had another hang ups while using hardware screen, but I don't think it is related.

KevinOConnor, 2021-03-19:
Sorry, but we don't use github to help diagnose problems with individual printers.  That said, some other process on your host computer is using too much system resources - you'll need to track that down and fix it.

-Kevin

Yury-MonZon, 2021-03-20:
Oh I see.
So it is definitely not Klipper's error? I couldn't get the reason from the log file.

If so - I'll deal with it myself. Just needed to confirm what's the problem.

I'll close the issue after your confirmation.

Thank you very much.

closed 2021-03-28 by KevinOConnor

MattByczko, 2021-09-11:
@Yury-MonZon  Did you ever sort this issue out?