Klipper3d/klipper #4066: Problems using FYSETC S6 with TMC2130 and FYSETC Mini12864 LCD

jonathanlundstrom, 2021-03-17:
Hi,

I'm running Klipper on most of my machines and I absolutely love it, so I thought that it would be easy to get it working on my cartesian printer (bedslinger) as well. But boy was I wrong.. The printer is running the following specs:

- FYSETC S6 control board
- FYSETC Mini12864 v2.1 LCD
- Genuine Watterott TMC2130 drivers

I've managed to get all the basics of the machine up and running, but there are two things that still no not work as they should. The first one is sensorless homing (stallGuard) and the second one is display output. The control board has integrated SPI and UART so stallGuard should work without any wires (it does on Marlin), but with Klipper my motors will simply slam into the side and continue making horrible noises while they loose steps. Even at values as low as `-64`. The screen will light up with the proper colors, but there is no actual output on it.

This is how my printer configuration looks right now:
```
########################################
# Fluidd settings
########################################
[virtual_sdcard]
path: ~/gcode_files

[display_status]

[pause_resume]

########################################
# Printer & board
########################################

[mcu]
serial: /dev/serial/by-id/usb-Klipper_stm32f446xx_3D0032000551383437373038-if00
restart_method: command

[printer]
kinematics: cartesian
max_velocity: 200
max_accel: 2000
max_z_velocity: 25
max_z_accel: 100

########################################
# X-axis
########################################

[stepper_x]
microsteps: 16
step_pin: PE11
dir_pin: PE10
enable_pin: !PE12
rotation_distance: 32
position_endstop: 0
position_max: 200
homing_speed: 30
homing_retract_dist: 0
endstop_pin: tmc2130_stepper_x:virtual_endstop

[tmc2130 stepper_x]
spi_bus: spi1
cs_pin: PE7
diag1_pin: PB14
interpolate: True
run_current: 0.750
hold_current: 0.500
stealthchop_threshold: 250
driver_SGT: -64

########################################
# Y-axis
########################################

[stepper_y]
microsteps: 16
step_pin: PD8
dir_pin: PB12
enable_pin: !PD9
rotation_distance: 40
position_endstop: 0
position_max: 200
homing_speed: 30
homing_retract_dist: 0
endstop_pin: tmc2130_stepper_y:virtual_endstop

[tmc2130 stepper_y]
spi_bus: spi1
cs_pin: PE15
diag1_pin: PB13
interpolate: True
run_current: 0.800
hold_current: 0.500
stealthchop_threshold: 250
driver_SGT: -64

########################################
# Z-axis
########################################

[stepper_z]
microsteps: 16
step_pin: PD14
dir_pin: !PD13
enable_pin: !PD15
rotation_distance: 8
position_min: -5
position_max: 200
endstop_pin: probe:z_virtual_endstop

[stepper_z1]
microsteps: 16
step_pin: PE6
dir_pin: !PC13
enable_pin: !PE5
rotation_distance: 8

[tmc2130 stepper_z]
spi_bus: spi1
cs_pin: PD10
diag1_pin: PA0
run_current: 0.550
hold_current: 0.550
stealthchop_threshold: 50

[tmc2130 stepper_z1]
spi_bus: spi1
cs_pin: PC14
diag1_pin: PA2
run_current: 0.550
hold_current: 0.550
stealthchop_threshold: 50

########################################
# Extruder
########################################

[thermistor slice_engeering]
temperature1: 20
resistance1: 613400
temperature2: 250
resistance2: 1591
temperature3: 500
resistance3: 101.1

[extruder]
microsteps: 16
step_pin: PD5
dir_pin: PD6
enable_pin: !PD4
rotation_distance: 7.710843373
nozzle_diameter: 0.400
filament_diameter: 1.750
heater_pin: PB3
sensor_type: slice_engeering
#sensor_type: EPCOS 100K B57560G104F
sensor_pin: PC0
control: pid
pid_Kp: 22.2
pid_Ki: 1.08
pid_Kd: 114
min_temp: 0
max_temp: 260

[tmc2130 extruder]
spi_bus: spi1
cs_pin: PD7
diag1_pin: PA3
run_current: 0.420
hold_current: 0.420
stealthchop_threshold: 0

########################################
# Heaters
########################################

[heater_bed]
heater_pin: PC8
sensor_type: ATC Semitec 104GT-2
sensor_pin: PC3
control: watermark
min_temp: 0
max_temp: 130

########################################
# Fans
########################################

[fan]
pin: PB15

[heater_fan e0_fan]
pin: PB0
heater: extruder
heater_temp: 50.0
fan_speed: 1.0

[heater_fan control_fan]
pin: PB4

########################################
# Bed leveling settings
########################################

[bltouch]
sensor_pin: ^PA0   # Pull-up (^ symbol) needed in open drain mode
control_pin: PA3
probe_with_touch_mode: True
# Some BLTouch V3 and many clones apparently require this, though mine didnt:
#pin_up_touch_mode_reports_triggered: False
x_offset: 30
y_offset: -3
z_offset: 1.30
speed: 40
samples: 1

#[bed_mesh]
#mesh_min: 29,0
#mesh_max: 200,200
#probe_count: 4,4

########################################
# EXP1 / EXP2 (display) pins
########################################

[board_pins]
aliases:
    # EXP1 header
    EXP1_1=PC9, EXP1_2=PA8,
    EXP1_3=PC11, EXP1_4=PD2,
    EXP1_5=PC10, EXP1_6=PC12,    # Slot in the socket on this side
    EXP1_7=PD0, EXP1_8=PD1,
    EXP1_9=<GND>, EXP1_10=<5V>,

    # EXP2 header
    EXP2_1=PA6, EXP2_2=PA5,
    EXP2_3=PC6, EXP2_4=PA4,
    EXP2_5=PC7, EXP2_6=PA7,      # Slot in the socket on this side
    EXP2_7=PB10, EXP2_8=<RST>,
    EXP2_9=<GND>, EXP2_10=<5V>

######################################################################
# Fysetc Mini 12864Panel v2.1 (with neopixel backlight leds)
######################################################################

[display]
lcd_type: uc1701
cs_pin: EXP1_3
a0_pin: EXP1_4
rst_pin: EXP1_5
contrast: 63
encoder_pins: ^EXP2_5, ^EXP2_3
click_pin: ^!EXP1_2
# Some micro-controller boards may require an spi bus to be specified:
spi_bus: spi1
# Alternatively, some micro-controller boards may work with software spi:
#spi_software_miso_pin: EXP2_1
#spi_software_mosi_pin: EXP2_6
#spi_software_sclk_pin: EXP2_2

[output_pin beeper]
pin: EXP1_1

[neopixel fysetc_mini12864]
pin: EXP1_6
chain_count: 3
color_order: RGB
initial_RED: 1.0
initial_GREEN: 0
initial_BLUE: 0

########################################
# RGB header
########################################

[output_pin blue]
pin: PB7

[output_pin red]
pin: PB6

[output_pin green]
pin: PB5

######################################################################
# Macros
######################################################################

# Z-safe homing that (1) moves the probe up 15mm (2) home XY (3) move to center of bed (4) home Z
[homing_override]
set_position_z: 0
axes: xyz
gcode:
    G90
    G1 Z30 F600
    G28 X0 Y0
    G1 X100 Y100 F3600
    G28 Z0

[gcode_macro PAUSE]
rename_existing: BASE_PAUSE
default_parameter_X: 0    # edit to your preferred park position
default_parameter_Y: 0    # edit to your preferred park position
default_parameter_Z: 180     # edit to your preferred park position
default_parameter_E: 1      # edit to your preferred retract length
gcode:
    SAVE_GCODE_STATE NAME=PAUSE_state
    BASE_PAUSE
    G91
    G1 E-{E} F2100
    G1 Z{Z}
    G90
    G1 X{X} Y{Y} F6000

[gcode_macro RESUME]
rename_existing: BASE_RESUME
default_parameter_E: 1      # edit to your preferred retract length
gcode:
    G91
    G1 E{E} F2100
    G90
    RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
    BASE_RESUME

[gcode_macro CANCEL_PRINT]
rename_existing: BASE_CANCEL_PRINT
gcode:
    TURN_OFF_HEATERS
    CLEAR_PAUSE
    SDCARD_RESET_FILE
    BASE_CANCEL_PRINT
```

Being able to home the motors is of course the most important thing, and the LCD is secondary, but I would like to solve them both if possible. I suspect it's some error with SPI.

I've gone ahead and attached the log file from when the error occurs. Although I was not able to run M112 as G28 X locks the processing until the command has finished.

Thank you so much for your help!

[klippy.log](https://github.com/KevinOConnor/klipper/files/6160058/klippy.log)

trevjonez, 2021-03-18:
do you have all 6 drivers installed? I am only  seeing 5 configured here. if you do have 6 installed you need to either remove the unused one, or add configuration to disable the CS pin on that spare driver.

https://github.com/trevjonez/klipper-configs/blob/9735389a1d514d5a1f17bae30958a07354bb257b/fysetc_s6_common.cfg#L120

jonathanlundstrom, 2021-03-18:
@trevjonez, no, I only have 5 drivers installed `(X, Y, Z, E, E1 (Z1)`).

Is there anything else that I'm doing wrong based on my config?
I added the lines to disable the pin anyway, but it didn't help.

BlackStump, 2021-03-18:
Try updating to the latest commit

trevjonez, 2021-03-18:
based on the tmc dump in your log I don't think it is an SPI issue. 

```
DUMP_TMC stepper_x
========== Write-only registers ==========
IHOLD_IRUN: 0008180f IHOLD=15 IRUN=24 IHOLDDELAY=8
TPWMTHRS:   00000021 TPWMTHRS=33
TPOWERDOWN: 00000000
PWMCONF:    00050480 PWM_AMPL=128 PWM_GRAD=4 pwm_freq=1 pwm_autoscale=1
COOLCONF:   00400000 sgt=-64
========== Queried registers ==========
GCONF:      00000004 en_pwm_mode=1
GSTAT:      00000001 reset=1
IOIN:       11000078 DCIN_CFG5=1 DRV_ENN_CFG6=1 DCO=1 VERSION=0x11
TSTEP:      000fffff TSTEP=1048575
XDIRECT:    00000000
MSCNT:      00000008 MSCNT=8
MSCURACT:   00f7000c CUR_A=12 CUR_B=247
CHOPCONF:   14028384 toff=4 hend=7 TBL=1 vsense=1 MRES=4(16usteps) intpol=1
DRV_STATUS: 800f0000 CS_ACTUAL=15 stst=1
PWM_SCALE:  00000000
LOST_STEPS: 00000000
```
Probably worth double checking all the jumpers on the PCB to make sure the diag pins are being connected to the endstop pins.

jonathanlundstrom, 2021-03-18:
@trevjonez, I've been running this setup for the last two years using Marlin (with sensorless homing) and I did not alter any hardware configuration before moving to Klipper. So unless Klipper requires a different hardware setup something else must be wrong :(

trevjonez, 2021-03-18:
oh oh I think I see what it is. you need to pull up and invert the diag pins.

here is one of my old config files that was the S6 with 2130's you can compare with.

https://github.com/trevjonez/klipper-configs/blob/MK3-Bear-1/fysetc_s6_bear.cfg

jonathanlundstrom, 2021-03-18:
Yes! That absolutely worked – the printer can now home properly!
Do you also have any ideas on how I can get my screen working?

As you can see from the image, the colors I'm setting are getting through, but nothing else shows on the screen. Surely EXP1 and EXP2 must be oriented correctly as I haven't changed them since Marlin. Also, the reset button works, as does the beeper, even if it's a little faint.

[image]

trevjonez, 2021-03-18:
possibly the usb voltage drop issue? https://forum.fysetc.com/d/188-fysetc-s6-5v-rail-drops-to-4-4v-when-attaching-usb/3

jonathanlundstrom, 2021-03-18:
Doesn't seem like it. Also worked somewhat good under Marlin but I know that there were a lot of SPI conflicts and the content on my screen always stuttered every now and then (problem when using >=5 2130 drivers). Perhaps this is also an issue on Klipper? :(

trevjonez, 2021-03-18:
if we assume your config is correct this definitely feels like a hardware issue. how long are your 10 pin cables? i've had not great luck with long cables on 12864 displays in the past.

jonathanlundstrom, 2021-03-18:
Could it be that it's running on the same `spi_bus` as the motors? I tried changing it to `spi2` yesterday and also to use software SPI but that entails a bunch of other errors where it complains that several pins are reserved by `spi2`.

The 10-pin cables are only about 150mm long so I don't think that should be an issue. They just wrap around the frame into a hole in the bottom of the machine and into the EXP connectors.

trevjonez, 2021-03-18:
I am more wondering if you do need to rotate the 10 min cables. i've always had to on my machines, but I just assumed it was due to cheap displays being assembled wrong. perhaps it is an fysetc + marlin thing?

the not in the example config even references its rotation.
https://github.com/KevinOConnor/klipper/blob/master/config/generic-fysetc-s6.cfg

usually I just take a box cutter and slowly shave off the tab on the cable to make rotating trivial.

trevjonez, 2021-03-18:
I am not sure, mostly trying to recall all the bumps I hit with my machines that use the S6 1.2 board. But I've never had that fysetc display, and never had issues with other spi displays with all 6 drivers installed (only 5 in use)

jonathanlundstrom, 2021-03-18:
I've experimented some more and it is definitely a SW issue. Everything but the text output works (the most important part) and sometimes I'll see the screen do a light flicker for a moment. I also tried inverting the connectors but that didn't work at all. Maybe @KevinOConnor knows something we dont? 😅

tanaes, 2021-03-18:
Intriguingly, I'm experiencing the same issue with my FYSETC mini12864 v2.1 display (as well as the BTT mini12864 clone thereof) on my Klipper-powered printers, using MKS GenL V1.0 boards and TMC2100 clone drivers (Artillery Sidewinder X1 and Genius). I get Neopixels lighting up and working, but no text display.

closed 2021-03-22 by github-actions[bot]

trevjonez, 2021-03-22:
FWIW if you legit think it is a software issue and fits this bit https://www.klipper3d.org/Contact.html#i-found-a-bug-in-klipper
its probably worth opening a new ticket for that issue specifically.

jonathanlundstrom, 2021-03-22:
I would like to state that I do believe this is a bug in Klipper. I've gone through the pin mapping, pin by pin and I'm unable to get the LCD to output any characters - only the beeper, NeoPixels and reset button works. The rest of the config is correct.

tanaes, 2021-03-22:
@jonathanlundstrom I've been tracking this down as well... Should have some more hard info later today. Are you in the Klipper Discord? Maybe we can connect there in the interim. I'm whoppingpochard.

jonathanlundstrom, 2021-03-23:
It does seem that some errors still persist in relation to the drivers, and surely that is also what is messing with the SPI for the LCD. Was running a 5h print during the night and when I woke up the printer had stopped due to SPI errors on the X axis. I'm not sure how to continue debugging this but I'm sure @KevinOConnor would have some wise input. I would really like to know if this is a user/config fault or actually a bug in Klipper itself.

My log was too big to attach so I went ahead and uploaded it here:
https://we.tl/t-PKClfIXTRW

Thanks for your continued support :)

jonathanlundstrom, 2021-03-25:
So, to conclude – running 5 TMC2130 drivers while using the Mini12864 LCD is simply not possible using Klipper. The SPI gets overloaded and there is not enough room for motors and screen to reliably function together. Really a board issue more than a Klipper issue I suppose. I ended up switching to TMC2209/2208 drivers which works good, and the screen now also works since there is no congestion on the SPI bus. Thanks for all your help!