Klipper3d/klipper #4071: mechaduino/Servo42a AttributeError: MCU_virtual_stepper instance has no attribute 'is_dir_inverted'

aaalex2021, 2021-03-18:
First of all, **Klipper** is really a great piece of software. After I got aware of it last summer, it went to number one of the projects I spend my spare time with, either from 3D-printer-tuning side as well as from "understanding how **Klipper** works" point of view.

Now, to my issue. I got some **MKS Servo42a Servo-Steppers** lately and thought it would be fun to run **Klipper** on them before installing them in a 3D printer. I saw that @KevinOConnor already invested some effort into **Mechanduino / SmartStepper / Servo42** support in the [work-mechaduino-20181205](https://github.com/KevinOConnor/klipper/tree/work-mechaduino-20181205) branch through [PR1038](https://github.com/KevinOConnor/klipper/pull/1038).

So, I followed the recipe in [Mechaduino.md](https://github.com/KevinOConnor/klipper/blob/work-mechaduino-20181205/docs/Mechaduino.md), check out that branch, compiled the firmware and flashed it to the Servo42a. (took me some days and I learned even more ....) and basic operation works! I can control an LED on the board as an output_pin. How cool is that? Log attached
[servo42-20210318-1542-klippy.log](https://github.com/KevinOConnor/klipper/files/6165610/servo42-20210318-1542-klippy.log)

Then I added the servo-stepper configuration, but that then already throws an error in klippy and I am stuck.
```
Error loading kinematics 'cartesian'
Traceback (most recent call last):
  File "/home/klipper/klipper-servo42a/klippy/toolhead.py", line 258, in __init__
    self.kin = mod.load_kinematics(self, config)
  File "/home/klipper/klipper-servo42a/klippy/kinematics/cartesian.py", line 151, in load_kinematics
    return CartKinematics(toolhead, config)
  File "/home/klipper/klipper-servo42a/klippy/kinematics/cartesian.py", line 14, in __init__
    for n in ['x', 'y', 'z']]
  File "/home/klipper/klipper-servo42a/klippy/stepper.py", line 239, in LookupMultiRail
    rail = PrinterRail(config)
  File "/home/klipper/klipper-servo42a/klippy/stepper.py", line 134, in __init__
    stepper = PrinterStepper(config)
  File "/home/klipper/klipper-servo42a/klippy/stepper.py", line 96, in __init__
    self.is_dir_inverted = mcu_stepper.is_dir_inverted
AttributeError: MCU_virtual_stepper instance has no attribute 'is_dir_inverted'
Config error
```
full log with error:
[servo42-20210318-1158-klippy.log](https://github.com/KevinOConnor/klipper/files/6165010/servo42-20210318-1158-klippy.log)

Okay. That PR is 2 years old, now, so my questions are:

- Does it make sense to invest more time here or is that branch abandoned?
- What triggers the AttributeError: ?
- Are there any intentions to bring the mechaduino branch to an up-to-date version?

Thank you so much for your help! And for this software!
-Alex

TazerReloaded, 2021-04-05:
Interesting find!
I have two S42Bs V2.0 from BigTreeTech, which I can only use in open loop configuration right now because the stock firmware uses a PID loop for position control which introduces a whole lot of problems with accuracy and timing.

There is a rewrite of the firmware in active development, the current solution is to get rid of the PID loop and execute the MCUs step pulses directly, and only apply positional correction when the sensor detects that the motor is out of step: https://github.com/CAP1Sup/Intellistep
It's in a very early stage, but first tests are promising.

If Klipper could talk to those drivers directly that would be even more awesome, but there seem to be some problems with heat if the drivers run on full current all the time, so some sort of current management might be necessary.

aaalex2021, 2021-04-07:
Hi.
As long as there is no development foreseen for the mechaduino branch, this topic can be closed.

closed 2021-04-07 by aaalex2021