Klipper3d/klipper #4081: [FR] X-axis vertical belt attachment offset compensation

chunter1, 2021-03-20:
I got the following feature request:
A huge number of 3D printers have the issue, that their x-axis belt is not parallel to the rail resulting in nonlinearity/dimension issues along the x-axis.
In the image below, the issue is marked yellow.
The closer the extruder carridge comes to the left pulley, the bigger the error.

[image]

Is a compensation for this systematic error already implemented and if not, i would love to see that feature since A LOT of users would benefit from it.
The equation for compensation is simple an the user just has to specify the vertical offset of the belt mount in relation to the ideal, parallel mounting point.
Calculation would be based on a simple right triangle.

Hywelmartin, 2021-03-20:
fix stupid designs with sw.. not the best idea.. but useful..

chunter1, 2021-03-20:
> fix stupid designs with sw.. not the best idea.. but useful..

I basically agree, but fixing it in hardware unfortunately isn't that easy (if you wanna prevent flex and other issues).
And a software-fix would be a clear improvement especially with large prints or prints that need good accuracy.

chunter1, 2021-03-20:
By the way... also the motion speed is non linear ;)

Sineos, 2021-03-20:
How do you want to reliably determine the needed compensation?

chunter1, 2021-03-20:
I tried to graph the error magnitude between the "wanted" and the "real" X-position of a system with 300 mm x-axis dimension and a vertical belt offset of 10 mm.
Its basically calculating the pothagoras.

[image]

Hywelmartin, 2021-03-20:
it's not really so easy. as the tension of the belt differs..

chunter1, 2021-03-20:
> it's not really so easy. as the tension of the belt differs..

Yes, the tention differs but it differs across the entire length of the belt and thus has no effect on the position.

KevinOConnor, 2021-03-22:
FYI, we don't track feature requests on github - see: https://www.klipper3d.org/Contact.html .  If you're currently working on implementing this, then that's fine.  Otherwise, this isn't the right place to raise it.

-Kevin

chunter1, 2021-03-23:
@KevinOConnor Ok, i understand.
So just in case somebody is interested and able to implement the feature, here is the math:

**Absolute position motion:**
O ... vertical offset of the belt mount in relation to ideal parallel position (see graphics in the first post)
X1 ... target "true" position
X ... stepper motion distance
[image]

**Relative position motion:**
O ... vertical offset of the belt mount in relation to ideal parallel position (see graphics in the first post)
X1 ... target "true" position
X2 ... actual "true" position
X ... stepper motion distance
[image]

So it's simply the Pythagoras that has to be applied.

dmbutyugin, 2021-03-29:
FWIW, I believe those formulas are not entirely accurate. In particular, the belt will be forced to stretch / will shrink as the toolhead moves about, which is not accounted for. And the stepper motion distance X will be in the units of the 'stretched belt'. One way to think about this is that X1 and O are in true mm, but X is in the teeth units, with one unit being only approximately equal to 2mm (with GT2 belt).

chunter1, 2021-03-29:
> FWIW, I believe those formulas are not entirely accurate. In particular, the belt will be forced to stretch / will shrink as the toolhead moves about, which is not accounted for. And the stepper [...]

Please post the corrected formulas.

dmbutyugin, 2021-03-29:
Well, if only I knew. Without any guarantees that this is actually correct and ignoring some boundary effects (like size of pulleys and finite length of the carriage), when moving from true x0 to x1, the commanded dX distance for the stepper motor:

dX = 2 * L * ( sqrt(x1^2+O^2) / (L + sqrt(x1^2 + O^2) + sqrt((L-x1)^2 + O^2)) - sqrt(x0^2+O^2) / (L + sqrt(x0^2 + O^2) + sqrt((L-x0)^2 + O^2)))

where L is the distance between the opposite pulleys. The additional terms in the denominators reduce the deviation, I believe.

chunter1, 2021-05-04:
Solution for me is using this X-axis belt offset part on thingiverse for mc CR10s pro:
https://www.thingiverse.com/thing:4845116

closed 2021-05-04 by chunter1