Klipper3d/klipper #4087: Make `print_stats` to be central to print control

ayufan, 2021-03-21:
I'm new to klipper, but really love it's design principles.

While changing my prior Marlin+Octoprint to Klipper+Octoprint+Moonraker/fluidd I noticed a bunch of disrepancy how different systems track the print happening, and thus how they track the print duration and its state.

There's currently a handy plugin `print_stats` that is used exclusively by `virtual_sdcard` to inject a printing state (with filament/duration tracking) into a dedicated object. However, this alone work only with `virtual_sdcard` G-codes. This raises some problems how we can track print start/pause/resume/finish/error in a consistent manner across different systems, given that klipper additionally offers `pause_resume` plugin.

It would help to have a single way of tracking a current print status in a consistent way, allowing to later define all parameters of print, including:
- filament used and current duration
- expected filament usage and remaining time
- tracking prints in a form a simple counter
- having a well defined entry/exit points for prints happening

I'm happy to make changes in either of these directions. My personal interest is being able to use concurrently `virtual_sdcard` and mostly `Octoprint`, but be able to implement a simple print tracking in form of [M78](https://marlinfw.org/docs/gcode/M078.html) gcode.

### Proposal 1

Unify and make a `print_stats` to be central place to control the current print progress regardless of method being used, defining a well defined state pattern that would be used automatically (by `virtual_sdcard`) or manually (by custom gcodes in Octoprint or other systems). Allowing, users to extend a well defined entry points with their own gcodes.

The idea with steps:

- introduce `START_PRINT`, `RESUME`, `PAUSE`, `CANCEL_PRINT `, `ERROR_PRINT` that are defined by `print_stats` that changes the internal state of `print_stats` as it is reflected by the current state machine
- make a proper events to be generated by system when these actions happens that can be interpreted by others
- make `virtual_sdcard` to not call `print_stats.note_*`, but rather execute `START_*` gcodes
- make `RESUME`, `PAUSE` and `CANCEL_PRINT` to be extended by `pause_resume` not defined

### Proposal 2

Allow `print_stats` to be used outside of `virtual_sdcard` and:

- introduce `STATS_START`, `STATS_PAUSE`, `STATS_COMPLETE`
- make `virtual_sdcard` to not call `print_stats.note_*`, but rather execute `STATS_*` gcodes
- make `pause_resume` to additionally call `STATS_START/PAUSE`

### Example usage

- extend `START_PRINT` and `FINISH_PRINT` with print history by a user gcode macros `.cfg` allowing to implement [M78](https://marlinfw.org/docs/gcode/M078.html)
- have a single entry point defined that is used by `virtual_sdcard`, where as there is not difference in tracking and behavior between different ways of running prints: an internal print (sdcard) or an external print (octoprint?)
- optionally inject `START_PRINT` and `FINISH_PRINT` on actions on heaters similarly to `Marlin`, where a set of state changes to heaters (with wait or not) can be interpreted as `START_PRINT/FINISH_PRINT`

### Reference

- fluidd calculating and showing print time and duration: https://github.com/cadriel/fluidd/blob/966ec8b2c2e00d51fe10d81673fc1a858e242e12/src/store/helpers.ts#L140, https://github.com/cadriel/fluidd/blob/966ec8b2c2e00d51fe10d81673fc1a858e242e12/src/store/printer/getters.ts#L78
- fluidd showing a progress: https://github.com/cadriel/fluidd/blob/966ec8b2c2e00d51fe10d81673fc1a858e242e12/src/store/printer/getters.ts#L105
- fluidd showing time estimates: https://github.com/cadriel/fluidd/blob/966ec8b2c2e00d51fe10d81673fc1a858e242e12/src/store/printer/getters.ts#L128

KevinOConnor, 2021-03-22:
FWIW, I agree the state tracking could be improved.

I'm not sure how your proposal 1 differs from your proposal 2.  (Unless you are suggesting proposal 2 is a subset of proposal 1.)  I don't see any harm in introducing macros that OctoPrint can call.  In that case, the macros should be new names though - as in PRINT_STATS_START, PRINT_STATS_FINISH, etc.  I don't see any harm in changing virtual_sdcard (and similar) to call those new macros.

FWIW, though, I get the impression the core development team has "moved on from OctoPrint".  Some of the interface features that can be accomplished via the newer UIs (mainsail, fluidd, etc.) are compelling and aren't plausible via OctoPrint.  I still use OctoPrint exclusively, but I suspect that wont last long.

-Kevin

ayufan, 2021-03-22:
@KevinOConnor 

I still continue using OctoPrint, but I see that the whole ecosystem very nicely works towards OctoPrint-less world. However, the problem still stays: a way to track the start and end of the print, with its progress in a consistent way across different tools (if possible).

I'm trying to figure out a way that would make it a defined and generic way, to which we could look all operations. I noticed the general pattern today to use:
- `START_PRINT`, `CANCEL_PRINT` and `END_PRINT` as indirectly mentioned in https://github.com/meteyou/mainsail/blob/c22d0c20403b61128d73ce35f6db6d88db360e8b/docker/config/printer.cfg.example#L145
- fluidd does not define that: https://docs.fluidd.xyz/configuration/initial_setup#macros
- I see that moonraker added a history tracking with a usage of `print_stats`: https://github.com/Arksine/moonraker/blob/f38cacd513fab59a625a9a19e0725e41f7e6f545/moonraker/components/history.py#L109. It simply looks over state change of `print_stats`

Maybe the best could be to define (and re-use the following keywords):

### Style 1

- `START_PRINT` defined by `print_stats`
- `END_PRINT` defined by `print_stats`
- `CANCEL_PRINT`, `PAUSE`, `RESUME` be defined by `print_stats`, and extended by `pause_resume`
- make `virtual_sdcard` as part of `M2x` call `START/CANCEL/END_PRINT`

### Style 2

Maybe this would be better?

- Make `pause_resume` to depend on `print_stats`
- `START_PRINT` define by `pause_resume`, call `print_stats` directly via `note_start`
- `END_PRINT` define by `pause_resume`, call `print_stats` directly via `note_complete`
- `PAUSE`/`RESUME`/`CANCEL_PRINT` as currently defined, calls `print_stats`
- make `virtual_sdcard` to depend on `pause_resume` instead of `print_stats`
- make `M2x` call `START/CANCEL/END_PRINT`

KevinOConnor, 2021-03-22:
The `START_PRINT` macro is recommended at https://www.klipper3d.org/Slicers.html#klipper-gcode_macro and means something different than what is discussed here.  (It is intended to be called from the Slicer.)

> Make pause_resume to depend on print_stats

We can load modules like the above by default, so order of import isn't important.

-Kevin

closed 2021-05-05 by github-actions[bot]