Klipper3d/klipper #4092: Can't compile STM32F446 code the second time

bruce356, 2021-03-22:
Just received my Fysetc S6 board and am having the same problem. The first firmware make and upload worked as expected. On the second attempt after "make clean"
make: *** No rule to make target 'scripts/buildcommands.py', needed by 'out/compile_time_request.o'. Stop.
The same error comes up when attempting to "make flash FLASH_DEVICE=/dev/ttyACM0" with a backup copy placed into the "out" folder.
After re-installing Klipper it works for the first firmware upload but not the second attempt.
Thanks regards - Bruce

https://github.com/KevinOConnor/klipper/issues/473
https://github.com/KevinOConnor/klipper/pull/3562

Sineos, 2021-03-22:
try `make distclean`

bruce356, 2021-03-22:
@Sineos thanks for the reply, tried your suggestion but get following errors 
src/i2ccmds.c:14:23: error: field ‘i2c_config’ has incomplete type
     struct i2c_config i2c_config;
                       ^
src/i2ccmds.c: In function ‘command_config_i2c’:
src/i2ccmds.c:23:23: warning: implicit declaration of function ‘i2c_setup’ [-Wimplicit-function-declaration]
     i2c->i2c_config = i2c_setup(args[1], args[2], addr);
                       ^
src/i2ccmds.c: In function ‘command_i2c_write’:
src/i2ccmds.c:35:5: warning: implicit declaration of function ‘i2c_write’ [-Wimplicit-function-declaration]
     i2c_write(i2c->i2c_config, data_len, data);
     ^
src/i2ccmds.c: In function ‘command_i2c_read’:
src/i2ccmds.c:48:5: warning: implicit declaration of function ‘i2c_read’ [-Wimplicit-function-declaration]
     i2c_read(i2c->i2c_config, reg_len, reg, data_len, data);
     ^
make: *** [Makefile:64: out/src/i2ccmds.o] Error 1

Sineos, 2021-03-22:
> Edit:
> You would need to run `make menuconfig` afterwards

bruce356, 2021-03-23:
@Sineos, tried as you suggest above, again thanks, running "make menuconfig" afterwards allowed for fresh reconfiguring of settings, but at the end of the firmware build process it would fail with the same original error.
Had no other choice but to start again from scratch, and now it is working, do not know why.
Regards - Bruce

closed 2021-03-23 by bruce356