Klipper3d/klipper #4097: error: Internal error in stepcompress

evolwun, 2021-03-23:
I am having an issue (more than a few times) where a print errors out. I get the followng:

```
stepcompress o=10 i=-1 c=1 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 141, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 175, in move
    self.toolhead.move([x, y, z + self.fade_target, e], speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 420, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 289, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 185, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
```

I am running ver v0.9.1-336 of Klipper on a BTT SKR 1.1 Pro. This has happened multiple times at varying times in a print. Fluidd displays the error below. I have tried a fresh build of FluiddPi it still occurs on many different models. klippy log is attached.
[klippy (2).log](https://github.com/KevinOConnor/klipper/files/6186956/klippy.2.log)

[image]

evolwun, 2021-03-30:
It happened again. The klippy.log is too big to attach. Here is the section with the error:

`
Stats 307487.9: gcodein=0 mcu: mcu_awake=0.008 mcu_task_avg=0.000011 mcu_task_stddev=0.000012 bytes_write=50892882 bytes_read=46240222 bytes_retransmit=9 bytes_invalid=0 send_seq=1339016 receive_seq=1339016 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0 freq=168000364 heater_bed: target=85 temp=85.2 pwm=0.012 sd_pos=2295159 pi_cpu: temp=39.2 print_time=612770.746 buffer_time=2.445 print_stall=0 extruder: target=245 temp=244.9 pwm=0.617 sysload=0.04 cputime=1435.927 memavail=763524
stepcompress o=9 i=-1 c=1 a=0: Invalid sequence
stepcompress o=9 i=-1 c=1 a=0: Invalid sequence
stepcompress o=9 i=-1 c=1 a=0: Invalid sequence
stepcompress o=9 i=-1 c=1 a=0: Invalid sequence
Internal error on command:"G1"
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/gcode.py", line 182, in _process_commands
    handler(gcmd)
  File "/home/pi/klipper/klippy/extras/gcode_move.py", line 141, in cmd_G1
    self.move_with_transform(self.last_position, self.speed)
  File "/home/pi/klipper/klippy/extras/bed_mesh.py", line 181, in move
    self.toolhead.move(split_move, speed)
  File "/home/pi/klipper/klippy/toolhead.py", line 420, in move
    self.move_queue.add_move(move)
  File "/home/pi/klipper/klippy/toolhead.py", line 186, in add_move
    self.flush(lazy=True)
  File "/home/pi/klipper/klippy/toolhead.py", line 175, in flush
    self.toolhead._process_moves(queue[:flush_count])
  File "/home/pi/klipper/klippy/toolhead.py", line 336, in _process_moves
    self._update_move_time(next_move_time)
  File "/home/pi/klipper/klippy/toolhead.py", line 289, in _update_move_time
    sg(sg_flush_time)
  File "/home/pi/klipper/klippy/stepper.py", line 185, in generate_steps
    raise error("Internal error in stepcompress")
error: Internal error in stepcompress
Transition to shutdown state: Internal error on command:"G1"
Dumping gcode input 0 blocks
gcode state: absolute_coord=True absolute_extrude=False base_position=[0.0, 0.0, 0.0, 29103.64390199172] last_position=[104.61, 156.783, 1.96, 30165.356323990327] homing_position=[0.0, 0.0, 0.0, 0.0] speed_factor=0.0166666666667 extrude_factor=0.94 speed=5.0
Virtual sdcard (2294362): ' Y157.554 E0.2513\r\nG1 X104.251 Y150.210 E0.3517\r\nG0 F2700 X104.251 Y150.110\r\nG0 F6000 X103.841 Y149.700\r\nG0 F6000 X103.771 Y149.630\r\nG1 F2700 X110.232 Y149.630 E0.3094\r\nG1 X110.232 Y157.499 E0.3769\r\nG1 X109.697 Y158.034 E0.0362\r\nG1 X103.771 Y158.034 E0.2838\r\nG1 X103.771 Y149.730 E0.3977\r\nG0 F2700 X103.771 Y149.630\r\nG0 F6000 X103.361 Y149.220\r\nG0 F6000 X103.291 Y149.150\r\n;TYPE:WALL-OUTER\r\n;WIDTH:0.480\r\nG1 F1500 X110.712 Y149.150 E0.3554\r\nG1 X110.712 Y157.698 E0.4094\r\nG1 X109.896 Y158.514 E0.0553\r\nG1 X103.291 Y158.514 E0.3163\r\nG1 X103.291 Y149.250 E0.4437\r\nG0 F1500 X103.291 Y149.150\r\nG0 F6000 X106.767 Y150.436\r\nG0 F6000 X106.860 Y150.470\r\n;TYPE:FILL\r\n;WIDTH:0.480\r\nG1 F4800 X106.860 Y157.194 E0.3220\r\nG1 X109.238 Y157.193 E0.1139\r\nG1 X109.390 Y157.105 E0.0084\r\nG1 X109.392 Y152.892 E0.2018\r\nG1 X105.194 Y150.469 E0.2321\r\nG1 X109.365 Y150.469 E0.1998\r\nG1 X104.610 Y153.214 E0.2630\r\nG1 X104.611 Y157.194 E0.1906\r\nG1 X105.319 Y157.192 E0.0339\r\nG1 X104.696 Y156.833 E0.0344\r\nG0 F4800 X104.610 Y156.783\r\nG1 F1800 E-0.8000\r\n'
Upcoming (2295386): 'G1 F300 Z1.960\r\nG0 F6000 X79.287 Y130.899\r\nG0 F6000 X79.218 Y130.828\r\n;TYPE:WALL-INNER\r\n;WIDTH:0.480\r\nG1 F300 Z1.160\r\nG1 F1800 E'
Reactor garbage collection: (307464.609131357, 307239.345559828, 304781.591963162)
Internal error on command:"G1"
MCU 'mcu' shutdown: Command request
clocksync state: mcu_freq=168000000 last_clock=102945080882431 clock_est=(307459.360 102940280445762 168000363.459) min_half_rtt=0.000135 min_rtt_time=307138.210 time_avg=307459.360(849.139) clock_avg=102940280445762.516(142655706520.975) pred_variance=13716836.987
Dumping serial stats: bytes_write=50892965 bytes_read=46240391 bytes_retransmit=9 bytes_invalid=0 send_seq=1339023 receive_seq=1339023 retransmit_seq=2 srtt=0.001 rttvar=0.000 rto=0.025 ready_bytes=0 stalled_bytes=0
Dumping send queue 100 messages
Sent 0 307485.964203 307485.964203 6: seq: 1b, get_clock
Sent 1 307485.996298 307485.996298 58: seq: 1c, queue_step oid=11 interval=30726435 count=1 add=0, queue_step oid=12 interval=30726435 count=1 add=0, queue_step oid=13 interval=189504 count=4 add=8418, queue_step oid=13 interval=243659 count=1 add=0, set_next_step_dir oid=11 dir=1, queue_step oid=11 interval=12717376 count=1 add=0, set_next_step_dir oid=12 dir=1, queue_step oid=12 interval=12717376 count=1 add=0
`

Gilabite, 2021-03-30:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

evolwun, 2021-03-30:
> https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

That is what lead me to create an issue. If there is more info needed, I am happy to add it.

KevinOConnor, 2021-03-31:
An "internal error" like this is generally the result of a software defect.  Unfortunately, I have not had a chance to look at this closer.

Separately, there was a similar report at #3980.

-Kevin

evolwun, 2021-04-09:
Happened again on a completely different mcu (same Pi and same stepper drivers), I swapped over to an SKR 1.4 Turbo. Could this be something in the gcode?

[klippy.log](https://github.com/KevinOConnor/klipper/files/6288657/klippy.log)

KevinOConnor, 2021-04-11:
I've done an investigation of this issue.  My best guess is the error is due to an obscure numerical stability issue in the code.  I've made a code change that I think may help (commit dce9971a).  I recommend updating to the very latest code.

Unfortunately, I'm unable to reproduce the problem, so I can't be sure the above will fix the issue.  My best guess is that you are encountering this obscure error more frequently than others because you have a "round" step size (.010mm) that is more likely to perfectly overlap with positions found in gcode commands.  (And thus more likely to run into a numerical stability issue in the code.)  Other reporters of this issue (#3806 and #4122) have the same "perfect" step size.

-Kevin

closed 2021-05-15 by github-actions[bot]