Klipper3d/klipper #4100: Ender 6 / Installed BLTouch / unkwown pin chip name "probe"

onlycoop, 2021-03-23:
===== Config file =====
[virtual_sdcard]
path = ~/gcode_files

[display_status]

[pause_resume]

[stepper_x]
step_pin = PB8
dir_pin = PB7
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA5
position_endstop = 260
position_max = 260
homing_speed = 50

[stepper_y]
step_pin = PC2
dir_pin = !PB9
enable_pin = !PC3
microsteps = 16
rotation_distance = 40
endstop_pin = ^PA6
position_endstop = 260
position_max = 260
homing_speed = 50

[stepper_z]
step_pin = PB6
dir_pin = PB5
enable_pin = !PC3
microsteps = 16
rotation_distance = 8
endstop_pin = probe:z_virtual_endstop
position_min = -5
position_max = 400
	[safe_z_home]
	home_xy_position: 150.7, 137
	speed: 100
	z_hop: 10
	z_hop_speed: 5
	[bltouch]
	sensor_pin: ^PB1
	control_pin: PB0
	x_offset: 7
	y_offset: -67
	z_offset: 2.4
	speed: 3.0
	[bed_mesh]
	speed: 100
	mesh_min: 10, 10
	mesh_max: 239, 239
	algorithm: bicubic
	probe_count: 5,5

[extruder]
max_extrude_only_distance = 1000.0
step_pin = PB4
dir_pin = !PB3
enable_pin = !PC3
microsteps = 16
rotation_distance = 7.6
nozzle_diameter = 0.400
filament_diameter = 1.750
heater_pin = PA1
sensor_type = ATC Semitec 104GT-2
sensor_pin = PC5
control = pid
pid_kp = 26.949
pid_ki = 1.497
pid_kd = 121.269
min_temp = 0
max_temp = 300

[heater_bed]
heater_pin = PA2
sensor_type = EPCOS 100K B57560G104F
sensor_pin = PC4
control = pid
pid_kp = 327.11
pid_ki = 19.20
pid_kd = 1393.45
min_temp = 0
max_temp = 100

[fan]
pin = PA0

[filament_switch_sensor e0_sensor]
switch_pin = PA4

[mcu]
serial = /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
restart_method = command

[printer]
kinematics = corexy
max_velocity = 500
max_accel = 2000
max_accel_to_decel = 2000
max_z_velocity = 10
max_z_accel = 100

[gcode_macro PAUSE]
rename_existing = BASE_PAUSE
default_parameter_e = 1.7
gcode = 
	{% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
	{% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
	{% set max_z = printer.toolhead.axis_maximum.z|float %}
	{% set act_z = printer.toolhead.position.z|float %}
	{% if act_z < (max_z - 2.0) %}
	{% set z_safe = 2.0 %}
	{% else %}
	{% set z_safe = max_z - act_z %}
	{% endif %}
	SAVE_GCODE_STATE NAME=PAUSE_state
	BASE_PAUSE
	G91
	G1 E-{E} F2100
	G1 Z{z_safe} F900
	G90
	G0 X{x_park} Y{y_park} F6000
	G91

[gcode_macro RESUME]
rename_existing = BASE_RESUME
default_parameter_e = 1
gcode = 
	G91
	G1 E{E} F2100
	G90
	RESTORE_GCODE_STATE NAME=PAUSE_state MOVE=1
	BASE_RESUME

[gcode_macro CANCEL_PRINT]
rename_existing = BASE_CANCEL_PRINT
gcode = 
	TURN_OFF_HEATERS
	CLEAR_PAUSE
	SDCARD_RESET_FILE
	BASE_CANCEL_PRINT
=======================
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 146, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    m.add_printer_objects(config)
  File "/home/pi/klipper/klippy/toolhead.py", line 597, in add_printer_objects
    config.get_printer().add_object('toolhead', ToolHead(config))
  File "/home/pi/klipper/klippy/toolhead.py", line 259, in __init__
    self.kin = mod.load_kinematics(self, config)
  File "/home/pi/klipper/klippy/kinematics/corexy.py", line 108, in load_kinematics
    return CoreXYKinematics(toolhead, config)
  File "/home/pi/klipper/klippy/kinematics/corexy.py", line 14, in __init__
    stepper.LookupMultiRail(config.getsection('stepper_z')) ]
  File "/home/pi/klipper/klippy/stepper.py", line 371, in LookupMultiRail
    rail = PrinterRail(config)
  File "/home/pi/klipper/klippy/stepper.py", line 269, in __init__
    self.add_extra_stepper(config)
  File "/home/pi/klipper/klippy/stepper.py", line 347, in add_extra_stepper
    mcu_endstop = ppins.setup_pin('endstop', config.get('endstop_pin'))
  File "/home/pi/klipper/klippy/pins.py", line 256, in setup_pin
    pin_params = self.lookup_pin(pin_desc, can_invert, can_pullup)
  File "/home/pi/klipper/klippy/pins.py", line 239, in lookup_pin
    pin_params = self.parse_pin(pin_desc, can_invert, can_pullup)
  File "/home/pi/klipper/klippy/pins.py", line 224, in parse_pin
    raise error("Unknown pin chip name '%s'" % (chip_name,))
error: Unknown pin chip name 'probe'
webhooks client 3045606984: New connection
webhooks client 3045606984: Client info {'program': 'Moonraker', 'version': 'v0.4.1-85-gf0a6620'}

hello just installed a bltouch onto the ender. the klipper firmware was working fine before we started now this error comes up in fluild.local.

Unknown pin chip name 'probe'
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Printer is halted

onlycoop, 2021-03-23:
Also the we are connected to the motherboard not the second little board, when we had it plugged in there, the z stepper would hum

BlackStump, 2021-03-23:
[Help it doesn't work](https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work)

closed 2021-03-26 by github-actions[bot]