Klipper3d/klipper #4101: Calibrated extruder but mcu crashes when i extrude afterwards

MrNoob0, 2021-03-23:
Hello i just installed klipper yesterday and i needed to calibrate my extruder, but when i extrude afterwards it just crashes

This is the error i got in octoprint:
Recv: // MCU 'mcu' shutdown: Timer too closeRecv: // This often indicates the host computer is overloaded. CheckRecv: // for other processes consuming excessive CPU time, high swapRecv: // usage, disk errors, overheating, unstable voltage, orRecv: // similar system problems on the host computer.Recv: // Once the underlying issue is corrected, use theRecv: // "FIRMWARE_RESTART" command to reset the firmware, reload theRec

here is the logs
[klippy.log](https://github.com/KevinOConnor/klipper/files/6192784/klippy.log)

MrNoob0, 2021-03-23:
I just tried to start a print and i got this in octoprint

[image]

Gilabite, 2021-03-23:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

MrNoob0, 2021-03-23:
yup i just checked everything and i forgot to put the mcu section in the config files. Sorry for wasting your time

closed 2021-03-23 by MrNoob0