Klipper3d/klipper #4103: ABL issues with Z endstop

Fabrizio3, 2021-03-24:
first and foremost, i dont believe i need a log file for this.
so yesterday i installed a DIY probe and everything was smooth as butter, but i have a problem, it doesnt matter what value i throw at the probe offset or Z endstop position (which ive read in some issues should be the same) the nozzle is always .5mm above the bed (the probe Z offset which for some reason its not taking into account at all) im running it with virtual Z endstop as its connected directly to the Z endstop (its also a microswitch) and ive also read its buggy
TLDR is there a way to fix that? PROBE command reads out at z=0 but when i home to the endstop its at -.515, i need it to be 0 on the endstop and 0 on the probe (taking in the offset) but it just doesnt, maybe im having a brain fart lol

Fabrizio3, 2021-03-24:
its like it's using the Z probe offset as the Z endstop offset, but i want them totally separate, i put the probe on the printer (allen key probe) and measure the bed, take it out and use the endstop as the Z0, but its not letting me

Fabrizio3, 2021-03-24:
Z probe offset = 0 --->stepper_z:0.000000 on everything (GET_POSITION command)
Z probe offset = 0.515 --->stepper_z:0.554000 again, even when changing Z endstop pos around it's like the G28 Z0 is trying to use the probe every time and not behave like a normal endstop
in my mind having them separate is the normal thing, you do G28 Z0 and that uses the endstop offset which would be "0", then you do the probe command and it triggers the Probe Offset "-0.515"

Gilabite, 2021-03-24:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

Fabrizio3, 2021-03-24:
> 
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

yep, need to look trough the whole documentation instead of googling random bits that dont work, hope i can fix it! probably did something stupid that it said in the documentation not to do and i missed it

Fabrizio3, 2021-03-24:
virtual pin nullifies Z endstop and exclusively uses the probe, which is not retractable in my case so i'd need to home Z, then remove it and continue printing, to have both independently in klipper you need to have it in separate pins, which my anet a8 doesn't have lol, so that's it, only solution i can imagine is maunally having the allen key probe be the same height as the nozzle so it's the same as Z endstop, but then Z would trigger before, so i guess that's a nope sadly

closed 2021-03-24 by Fabrizio3