Klipper3d/klipper #4104: ClockSync issue when using CAN

scotthsl, 2021-03-25:
Hi Kevin,
Recently I was reading the source code of klippy to understand its clock synchronization mechanism, but it is still a work in progress.
 Then I found a problem with the handle_message(struct serialqueue *sq, double eventtime, int len) function in the klippy/chelper/serialqueue.c file:

```
// Process a well formed input message
static int
handle_message(struct serialqueue *sq, double eventtime, int len)
{
    // Calculate receive sequence number
   ...

    // Check for pending messages on notify_queue
    ...

    // Process message
    if (len == MESSAGE_MIN) {
        // Ack/nak message
        ...
    } else {
        // Data message - add to receive queue
        struct queue_message *qm = message_fill(sq->input_buf, len);
        qm->sent_time = (rseq > sq->retransmit_seq
                         ? sq->last_receive_sent_time : 0.);
        qm->receive_time = get_monotonic(); // must be time post read()

        qm->receive_time -= sq->baud_adjust * len;  // here is my concern

        list_add_tail(&qm->node, &sq->receive_queue);
        must_wake = 1;
    }

    ...
}
```
You would use `baud_adjust` to correct the receive_time of respond. but `baud_adjust` is not necessary when using CAN bus and by reading the source code I found that it currently only works for UART interface. so I would like to ask if `baud_adjust` is 0 for CAN bus does it impact the clock synchronization in CAN bus scenario?

My machine contains a total of 7 MCUs, all controlled via CAN bus. 5 MCUs control the linear modules, 1 MCU controls Tool Head and 1 MCU controls hot bed. Currently it prints properly, but the model will occasionally be misaligned. I am trying to figure out how to solve this problem, and will try to optimize the clock synchronization mechanism in CAN bus scenario after I fully understand the clock management system of klipper.

Thanks
Scott

KevinOConnor, 2021-03-25:
If you think there is an issue with the code, please start by attaching the full Klipper log file from the incident: https://www.klipper3d.org/Contact.html

The baud_adjust setting is used to account for time spent transmitting data on slow UART lines.  (Specifically, we can use knowledge of the time it takes to transmit each byte to make more accurate clock estimates.)  It's disabled on the other protocols (USB, CAN) because these higher speed buses don't have as predictable timing and their higher speed makes it unnecessary.  The code cited above looks correct - if baud_adjust is zero then we use the raw system time.

If you think there are clock synchronization issues - a good place to start is the mcu frequency estimation found in the log data.  The `./scripts/graphstats.py -f` tool is useful for analyzing that info - there is some info in docs/Debugging.md on setting up the graph tools.

-Kevin

scotthsl, 2021-03-29:
Thanks for your advice！
And sorry for asking an unqualified question. will further analysis it!

Thanks
Scott

closed 2021-05-12 by github-actions[bot]