Klipper3d/klipper #4109: QUERY_ENDSTOPS not accepting custom endstop assignements

TheHackerArtist, 2021-03-26:
[klippy_2021_03_26_B.log](https://github.com/KevinOConnor/klipper/files/6213750/klippy_2021_03_26_B.log)

I Klipperized a Duet 2 w/ Duex 5. I am at the stage of testing the endstops are properly triggered. For wire management purposes I have did not follow the "expeted" motor mapping and endstop mapping. I have attached the log file containing my printer configurations ( there are two cfgs present, explained further down) but ill add more info here for clarity along with links to the Duet 2, Duex 5 pin assignements and naming.
```

## Drivers
# Here are the pins for the 10 stepper drivers supported by a Duet2 board
# | Drive Num | Drive Ref. | Current |  DIR pin |  STEP pin |  ENDSTOP pin |  SPI EN pin |
# |-----------|------------|---------|----------|-----------|--------------|-------------|
# |    0      | X          | -       |  PD11    |  PD6      |  PC14        |  PD14       |
# |    1      | Y          | X       |  PD12    |  PD7      |  PA2         |  PC9        |
# |    2      | Z          | X1      |  PD13    |  PD8      |  PD29        |  PC10       |
# |    3      | E0         | E0      |  PA1     |  PD5      |  PD10        |  PC17       |
# |    4      | E1         | -       |  PD9     |  PD4      |  PC16        |  PC25       |
# |    5      | E2         | Y       |  PD28    |  PD2      |  PE0*        |  PD23       |
# |    6      | E3         | -       |  PD22    |  PD1      |  PE1*        |  PD24       |
# |    7      | E4         | -       |  PD16    |  PD0      |  PE2*        |  PD25       |
# |    8      | E5         | Z1      |  PD17    |  PD3      |  PE3*        |  PD26       |
# |    9      | E6         | Z       |  PC0     |  PD27     |  PA17*       |  PB14       |
```
[Duex 2 pinout and naming](https://d17kynu4zpq5hy.cloudfront.net/igi/duet3d/4k56ZjaaAHrTJI6Z.huge)
[Duex 5 pinout and naming](https://d17kynu4zpq5hy.cloudfront.net/igi/duet3d/3wbm6g2qRUBfSHgw.huge)
[Detailed ATSAM pin mapping](https://duet3d.dozuki.com/Wiki/Duet_2_Pinout_table)

The table above is a summary of how I wired the stepper motors and their corresponding naming standards and associated pins, its a modified table from the standard configs pulled from github. Now I want to assign pins PA2 [X], PE0 [Y], and PA17 [Z] for my endstop such that I can wire them as per the image below - note Y is on the right and Z on the left at the bottom of the image on the duex 5 and that the steppers aren't all plugged in at this time:

[image]

Now when issuing QUERY_ENDSTOPS, I cannot get this custom setup to work, only the X axis seems to be working. Does the QUERY_ENDSTOPS expect the normal configuration of endstop only or is updated with custom mapping of the endstop? See attached image of the terminal - the first query had no endstops physically triggered, the second query has all endstop physically triggered - Only the X axis responds approprietely.

[image]

First query
[image]

Second query
[image]

Seems like a bug. For the record, I have succesfully mapped and triggered the X, Y, Z endstop and rewired them physically according to pins PD14, PC9 and  PC10 respectively ( expected setup). That is why you will find two sets of endstop pin assignement in the attached .log file.

TheHackerArtist, 2021-03-26:
Ah! Turns out there's a little comment under the table in the example configuration file stating that the endstops on the duex don't work. I am not sure I understand why that is, it would be nice addition if they were. But I guess this isnt a bug but more of a design choice so this thread is more of a feature request than a bug in that case.

FHeilmann, 2021-03-27:
The reason they don't work is that they are not directly hooked up to the MCU, but instead are used via the SX1509 i²c to GPIO expander. This makes their implementations eparate from regular endstops, and I haven't managed to figure it out so far.

closed 2021-05-10 by github-actions[bot]