Klipper3d/klipper #4115: Get MCU shutdown error in 3 consecutive print

jonathanoeijoeng, 2021-03-28:
Hi,

I got error MCU shutdown in my last 3 consecutive print on the same gcode file. I have printed this file many time before without any problem.
This is 3.5 hours printing and got MCU shutdown error in around 30 mins before completed time.
I use Ender 3 v2 with 4.2.2 stock mainboard. 

Thanks

[image]

[klippy.txt](https://github.com/KevinOConnor/klipper/files/6216686/klippy.txt)

Gilabite, 2021-03-28:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

KevinOConnor, 2021-03-28:
FWIW, you should read the FAQ entry on this: https://www.klipper3d.org/FAQ.html#i-keep-getting-random-lost-communication-with-mcu-errors

-Kevin

closed 2021-03-31 by github-actions[bot]