Klipper3d/klipper #4118: "Serial connection closed": Klipper/OctoPrint cannot connect on a Ender 3 v2 pro

LSaldyt, 2021-03-28:
I've tried quite a few things, but after installing Klipper on my Ender3 v2 Pro (this [one](https://www.amazon.com/Creality-Printer-Printing-Function-220x220x250mm/dp/B07FFTHMMN/ref=sr_1_3?dchild=1&keywords=ender+3+v2+pro&qid=1616906195&sr=8-3)), I've had no luck in getting it working. I used [these](https://www.reddit.com/r/klippers/comments/kj2h5r/stepbystep_guide_for_ender_3_v2_klipper_w_bltouch/) instructions as my main starting point.

From OctoPrint, after sending `status`, I get:
```
Recv: // Printer is not ready
Recv: // The klippy host software is attempting to connect.  Please
Recv: // retry in a few moments.
Recv: // Klipper state: Not ready
Recv: !! Printer is not ready
```
Important part of `klippy.log`:
```
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```

Here are my specific error logs and settings: 
[klippy.log](https://github.com/KevinOConnor/klipper/files/6216876/klippy.log)
[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/6216874/printer.cfg.txt)
[octoprint.log](https://github.com/KevinOConnor/klipper/files/6216875/octoprint.log).
[image]

Physically, I'm connecting the raspberry pi to the ender 3 via a USB cable from a USB 3.0 port to the front USB on my printer.
My firmware was flashed manually with an SD card.
This is the serial ID I'm using:
```
pi@octopi:~/klipper $ ls /dev/serial/by-id/usb-1a86_USB_Serial-if00-port0 
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
```

LSaldyt, 2021-03-28:
Oh, and I tried `baud=250000`, too. Current settings reflect what OctoPrint defaulted too.

LSaldyt, 2021-03-28:
"Invalid" how? You don't know how to fix this? I've provided tons of information on why your software is broken!

LSaldyt, 2021-03-28:
@KevinOConnor

wesk-ms, 2021-03-28:
Instead of being rude, try reading the prior post...

"We use github to share the results of work done to improve Klipper. We don't use github for requests. ( **In particular, we don't use github for feature requests, to answer questions, nor to help diagnose problems with a printer.)**
In the future, please follow the directions at: https://www.klipper3d.org/Contact.html".

LSaldyt, 2021-03-28:
I've read that link. It seems that the developers of Klipper simply don't want to assist in fixing their software?

wesk-ms, 2021-03-29:
Respectfully @LSaldyt...

First, your issue is most likely not a problem in the Klipper code that needs to be fixed.

Second, the Klipper developers have simply chosen not to use github for support requests.  It's just that simple.  There are links in the docs to the Klipper Discourse forum and the Klipper Discord chat where support requests like yours are welcomed.

closed 2021-04-07 by KevinOConnor

lowtolerance, 2021-05-02:
No, they simply don’t want to pollute discussion with issues that result from operator errors. This isn’t a problem with klipper, it’s a problem with you.