Klipper3d/klipper #4122: Internal error on command:"G1"

clsferguson, 2021-03-29:
[klippy.zip](https://github.com/KevinOConnor/klipper/files/6218601/klippy.zip)

Strange error, found the same issue on a google search of the error. The issue was resolved in a bugfix i think i understood. But it appears to be in my klippy.log

I'm going to reboot my printer and run the same print and see if it happens again. But i thought you may like to take a look.

jakep82, 2021-03-29:
You're running a version of Klipper that's more than 4 months old.  You should update Klipper before attempting the print again.

clsferguson, 2021-03-29:
I will look into updating. It hasn’t been causing much grief, just this one time.

For what its worth, I turned off my printer and turned it back on and ran the same gcode file. It is almost done now.

KevinOConnor, 2021-04-11:
As discussed in #4097, a change was made today for a similar issue.  I recommend updating to the very latest version of the code.

-Kevin

KevinOConnor, 2021-05-06:
I'm going to close this in favor of the existing #4097 .

-Kevin

closed 2021-05-06 by KevinOConnor