Klipper3d/klipper #4125: LPC1769 Serial connection does not work.

pmlody, 2021-03-30:
There is some problem with UART Serial communication.
I have valid configuration RPI <-> UART. It does work at marlin - device /dev/ttyAMA0 at 115200 and 250000 baud.
When I flash board with Klipper USB firmware, there is valid connection and everything is working.
But if I flash board with Klipper Serial firmware I only got:
```
Timeout on connect
Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
```
in klippy.log
I am certain there is no problem with connection between RPI and SKR board - same UART config works with RPI Zero and RPI 2B+ when using Marlin firmware. There must be some issue in firmware with serial connection itself. I have checked several historical version, from commit c83688b7bc7fcbc1a413d5707d11a2d2b674d48d till master and every firmware returns same error. I also checked 115200 baud and 250000 baud.
I have proper pronter.cfg:
```
[mcu]
serial: /dev/ttyAMA0
restart_method: command
baud: 250000
```
Same error with/without restart_method and baud.

Gilabite, 2021-03-30:
https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

pmlody, 2021-04-02:
[klippy.log](https://github.com/KevinOConnor/klipper/files/6249614/klippy.log)
I have added klipper.log file.

closed 2021-04-07 by KevinOConnor

IncredibleHohol, 2021-05-01:
> There is some problem with UART Serial communication.
> 
> I have valid configuration RPI <-> UART. It does work at marlin - device /dev/ttyAMA0 at 115200 and 250000 baud.
> 
> When I flash board with Klipper USB firmware, there is valid connection and everything is working.
> 
> But if I flash board with Klipper Serial firmware I only got:
> 
> ```
> 
> Timeout on connect
> 
> Wait for identify_response
> 
> Traceback (most recent call last):
> 
>   File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
> 
>     params = self.send_with_response(msg, 'identify_response')
> 
>   File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
> 
>     return src.get_response([cmd], self.default_cmd_queue)
> 
>   File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
> 
>     cmd_queue)
> 
>   File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
> 
>     raise error("Serial connection closed")
> 
> error: Serial connection closed
> 
> ```
> 
> in klippy.log
> 
> I am certain there is no problem with connection between RPI and SKR board - same UART config works with RPI Zero and RPI 2B+ when using Marlin firmware. There must be some issue in firmware with serial connection itself. I have checked several historical versions, from commit c83688b7bc7fcbc1a413d5707d11a2d2b674d48d till master and every firmware returns same error. I also checked 115200 baud and 250000 baud.
> 
> I have proper printer.cfg:
> 
> ```
> 
> [mcu]
> 
> serial: /dev/ttyAMA0
> 
> restart_method: command
> 
> baud: 250000
> 
> ```
> 
> Same error with/without restart_method and baud.

Did you solve your problem?

pmlody, 2021-05-01:
No.

totolook, 2021-05-02:
I same problem with rpi0w wit uart mode. 
* The port ttyAMA0 it's working in loopback (i jump the TX and RX pin)
* I connect the skr1.4T with FTDI usb  and i  receive gibberish output on my terminal console 
[image]
* I tested with the script ` $KLIPPY_EXEC /home/pi/klipper/klippy/console.py /dev/ttyAMA0` and this is the output: 
```bash 
pi@toto3d:~ $ klippy-env/bin/python /home/pi/klipper/klippy/console.py /dev/ttyAMA0

  This is a debugging console for the Klipper micro-controller.
  In addition to mcu commands, the following artificial commands are
  available:
    PINS  : Load pin name aliases (eg, "PINS arduino")
    DELAY : Send a command at a clock time (eg, "DELAY 9999 get_uptime")
    FLOOD : Send a command many times (eg, "FLOOD 22 .01 get_uptime")
    SUPPRESS : Suppress a response message (eg, "SUPPRESS analog_in_state 4")
    SET   : Create a local variable (eg, "SET myvar 123.4")
    STATS : Report serial statistics
    LIST  : List available mcu commands, local commands, and local variables
    HELP  : Show this text
  All commands also support evaluation by enclosing an expression in { }.
  For example, "reset_step_clock oid=4 clock={clock + freq}".  In addition
  to user defined variables (via the SET command) the following builtin
  variables may be used in expressions:
    clock : The current mcu clock time (as estimated by the host)
    freq  : The mcu clock frequency

==================== attempting to connect ====================
INFO:root:Starting serial connect
INFO:root:Timeout on connect
ERROR:root:Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
 cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
INFO:root:Timeout on connect
ERROR:root:Wait for identify_response
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/serialhdl.py", line 65, in _get_identify_data
    params = self.send_with_response(msg, 'identify_response')
  File "/home/pi/klipper/klippy/serialhdl.py", line 249, in send_with_response
    return src.get_response([cmd], self.default_cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 306, in get_response
    cmd_queue)
  File "/home/pi/klipper/klippy/serialhdl.py", line 241, in raw_send_wait_ack
    raise error("Serial connection closed")
error: Serial connection closed
^C
```
I tested also with cu and have same gibberish output
[image]

IncredibleHohol, 2021-05-02:
If u find answer, replay there pls. I'm tired of war with klipper.
In discord chat I got only one answer: "I have same build and it's work"