Klipper3d/klipper #4129: Klipper 0.9.1.-393 broke printer connection ("none type" has no attribute "get_position")

PM3Delta, 2021-04-01:
I think Klipper got accidentally broken during last nights update to 0.9.1-393.  I was on a previous version, just finished printing, then updated and got "Internal error during ready callback: 'NoneType' object has no attribute 'get_position'" and cant connect to the printer anymore.  Everything else seems up-to-date (see KIAUH screenshot).  Please find Klippy.log attached.  Let me know if I did something wrong or how I could revert back (sorry not very familiar with git commands).
[image]
[image]
[klippy.log](https://github.com/KevinOConnor/klipper/files/6241243/klippy.log)
[moonraker.log](https://github.com/KevinOConnor/klipper/files/6241245/moonraker.log)

henniero, 2021-04-01:
same probleme here on a anycubic trigorilla mcu (atmega2560).

Botasky556, 2021-04-01:
I just got the same error, it was working fine until I updated to the latest version of Klipper. Everything else is up to date as well. 
(MainsailOS, SKR 1.4 Turbo) 
[klippy.log](https://github.com/KevinOConnor/klipper/files/6241480/klippy.log)

Tekti, 2021-04-01:
If you need to start your printing, you can connect to rpi using for example WinSCP and modify file: /home/pi/klipper/klippy/extras/gcode-move.py. You need to comment out Line 57 or remove it from the file. After that, restart the rpi and you should be abble to run your printer, while waiting for the official update. 

[image]

th33xitus, 2021-04-01:
Imho it's a bad advice to modify the local klipper files instead of just rolling back to an earlier commit if you are really in need of doing a print right now. 
Modifing any local klipper file will force you to revert (or stash and pop) that changes anyways before you are able to do a `git pull`.

Also a possible solution could be to disable the loading of a bed mesh temporarily and just print without if you really have to.

Anyways... the easiest AND cleanest solution would be to just revert back 2 commit with `cd ~/klipper && git reset --hard HEAD~2` and wait for the fix.

just my2cents

henniero, 2021-04-01:
thx th33xitus! I'm a total Linux noop. This comand was the one I searched for `git reset --hard HEAD~2`!
As the noop i am, i donwnloaded the complete klipper snapshot from two days ago and installed it manually. actually that messed everything up. now it's working fine with v0.9.1-391....

FrincuFPV, 2021-04-01:
same problem here. Ender 3v2 ( V4.2.2. All good last night.:(

FrincuFPV, 2021-04-01:
For the moment I did this (cd ~/klipper && git reset --hard HEAD~2 and wait for the fix.) 
Thanks to (th33xitus)

chicco83, 2021-04-01:
> same probleme here on a anycubic trigorilla mcu (atmega2560).

same for me

rbnhln, 2021-04-01:
Same problem at Anycubic i3 mega with trigorilla board. Checked out at previous commit and everything works fine.

closed 2021-04-01 by KevinOConnor

PM3Delta, 2021-04-01:
Thank you very very much for the speedy fix. Really appreciate your efforts.

PR #4131 (2021-04-01): bed_mesh:  lookup the toolhead object in "handle_connect"
This is a bugfix that resolves #4129.