Klipper3d/klipper #4139: Delta Manual Test_Z menu doesn't work.

al3ph, 2021-04-04:
Whilst trying to get a similar menu working, I've found that the one I was basing mine off of doesn't work either.

Selection using a normal click, reverts to the sixth entry, instead of the one you select.

long_click appears to have no effect, i.e. the gcode macro isn''t executed,  or the long_click event isn't set.

[klippy.log](https://github.com/KevinOConnor/klipper/files/6255315/klippy.log)

```
[menu __main __setup __calib __delta_calib_man __test_z]
type: input
name: Test Z: {['++','+','+.01','+.05','+.1','+.5','-.5','-.1','-.05','-.01','-','--'][menu.input|int]}
input: 6
input_min: 0
input_max: 11
input_step: 1
gcode:
    {%- if menu.event == 'long_click' -%}
        TESTZ Z={['++','+','+.01','+.05','+.1','+.5','-.5','-.1','-.05','-.01','-','--'][menu.input|int]}
    {%- endif -%}

```

mcmatrix, 2021-04-06:
Thank you for the feedback.
I'll check it.

al3ph, 2021-04-06:
No prob, only foungd it as I was trying to implement the same functionality for Z_ENDSTOP_CALIBRATE

mcmatrix, 2021-04-06:
Hi @al3ph, 
I cannot try it on my printer at the moment.
But it's meant to work like this.

- You start editing via click (cursor turns to asterisk)
- You select the right action
- And now, if you want to execute that action you need to make a long press by holding the click button down.
  The long press event will happen automatically and then you can release.
- You can select another action or execute the same.
- You will not exit from menu element edit before you're done.
- If you exit via single click and reenter it will start always with the 6th action selected.

Can you confirm that "long press" is not working this way for you?

closed 2021-04-06 by al3ph