Klipper3d/klipper #4152: Message Bombs on error

EddieGreen, 2021-04-09:
Details in previously submitted ticket

[klippy.log](https://github.com/KevinOConnor/klipper/files/6287005/klippy.log)

KevinOConnor, 2021-04-09:
As mentioned in the previous ticket, we choose not to use github for support.  Please delete this ticket and refrain from opening new tickets.

-Kevin

EddieGreen, 2021-04-09:
So, is this a bug or normal behaviour?

I don't know why you choose to respond this way. Please explain why

I've read other issues and the ticket protocol and all I've done is provide detail on two issues that warrant the same kind of responses as other posts. I don't need you to resolve my issues with the amps or sensors, just clear down the ticket(s) with a meaningful response.

KevinOConnor, 2021-04-10:
I don't know.  We don't use github as tool to find out.

-Kevin

closed 2021-04-10 by KevinOConnor