Klipper3d/klipper #4154: [Needs advice] -  Inplementing support for some underground Cortex-M MCUs.

TheCodeExorcist, 2021-04-09:
NOTICE: In the first place I am going to introduce quickly who I am and what I do to help you understand why i want to do something that may seem stupid or pointless to you in the first place. If you prefer to stick to the point just skip, i won't get upset ;-)

Hi,
I am a french IT professional who focuses on custom 3D printing solutions for individuals and professionals, tailored on customers requirements. Due to COVID, my customers are increasingly seeking for cheaper machines that can deliver good performance for most needs, which leads me to expand my offer with a machine produced at least in small batch.

With this in mind I am currently designing an affordable yet particularly effective boxed printer using a DW-G gantry variant, 30cm³ build volume, lightweight direct drive extruder(s), strain gauge powered bed leveling (the Smart Effector principle), AC heaters and many other nice features which will make it quite a nice deal. To ensure that the print speed and quality matches the potential of the hardware, the goal is to make it run Klipper out of the box and as the machine will probably make use of more than one MCU, Klipper will also be the only firmware available for it.

Here comes the point:
To propose complete and very powerful electronics for an attractive price, i have to use more underground MCUs. Among my best matches for this project came two HDSC (Huada Semiconductor) MCUs: the HC32F460 and the HC32F072, which are respectively based on the ARM Cortex M4 and M0+ architectures. These MCUs are very similar to their ST Microelectronics counterparts but they also differ from them in many ways so there is no binary compatibility between them (We're not talking about CKS ripoffs here xD). I've seen that Klipper seems to use the sandard CMSIS abstraction layer for ARM based MCUs but sadly i never used CMSIS and I don't know what may be required to add the support for a new MCU. On the KEIL website no CMSIS content is available for these MCUs, however there seems to be some content provided by HDSC on their website but as I don't know that much about CMSIS i don't really know what to try. Any advice, tips and tricks are warmly welcome.
Thanks for your help,
Robin.

TheCodeExorcist, 2021-04-10:
UPDATE:
On my way to success, I finally found a quite complete CMSIS documentation and can confirm that HDSC does provide everything needed. I'll look forward to see precisely what needs to be done and keep the community updated on my progress.

JustAnother1, 2021-04-10:
If HDSC provides "everything needed" then what else do you need?

TheCodeExorcist, 2021-04-11:
>  [data:image/png;base64,iVBORw0KGgoAAAANSUhEUgAAAEgAAABICAYAAABV7bNHAAAABHNCSVQICAgIfAhkiAAAACtJREFUeJztwQENAAAAwqD3T20ON6AAAAAAAAAAAAAAAAAAAAAAAAAAAD4MUUgAARy2AfAAAAAASUVORK5CYII=###24x24:true###][Image de pistage][https://github.com/notifications/beacon/APAXBSAMNCIXCIP7LCGB2E3TIDO6PA5CNFSM42VPKBZ2YY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOGC237TQ.gif]

Well, HDSC provides everything needed to get support for those MCUs in the CMSIS layer, but there is still some coding required to map HDSC's functions with Klipper functions, as it has been done for other MCUs in the src folder. I started to code on the basis of what has been done for the STM32 MCUs and the HDSC docs and examples. If I have any question regarding my fonction mapping work I think the use of this issue should be relevant.
Best regards,
Robin.

closed 2021-05-24 by github-actions[bot]

alexqzd, 2021-08-17:
@TheCodeExorcist did you managed to make it work? I’m trying to do the same with the HC32F460 and Marlin with PlatformIO

henryabra, 2021-10-05:
@TheCodeExorcist What was the resolution on this?