Klipper3d/klipper #4155: CR10Max sensor_type issues.

nxvash, 2021-04-09:
I am working on getting Klipper working for my CR-10 Max.  I have everything installed and working up until doing a status after making the printer.cfg.  I got a copy of someone else's printer.cfg who is running a max but he has upgraded his heater carts and thermistors.  I am getting the following error when I send a status command.

"Send: status
Recv: // MCU 'mcu' shutdown: ADC out of range
Recv: // This generally occurs when a heater temperature exceeds
Recv: // its configured min_temp or max_temp.
Recv: // Once the underlying issue is corrected, use the
Recv: // "FIRMWARE_RESTART" command to reset the firmware, reload the
Recv: // config, and restart the host software.
Recv: // Printer is shutdown
Recv: // Klipper state: Not ready
Recv: !! MCU 'mcu' shutdown: ADC out of range
Recv: ok"

Does anyone know what I need to put in for the "sensor_type" under extruder in the cfg?  I am attaching a copy of my klippy.log as well incase that helps.

[klippy.log](https://github.com/KevinOConnor/klipper/files/6287912/klippy.log)

nxvash, 2021-04-11:
get the software and everything here but don't you DARE come here looking for help.  You have to go elsewhere to get support for the program I developed.  Makes total fucking sense.

BlackStump, 2021-04-11:
So let me get this straight, you have got your knickers in a knot and spitting the dummy because you have to go somewhere else then the Developers Git for software support, that you yourself have Contributed nothing to.
What gives you the right and makes you so special?
You come across as ungrateful, self centered sod.
I hope Kevin deletes this thread because that is all it deserves.

nxvash, 2021-04-11:
> So let me get this straight, you have got your knickers in a knot and spitting the dummy because you have to go somewhere else then the Developers Git for software support, that you yourself have [...]

And you sound like a douchebag who only posted to see his words on the screen.  I would say that you prolly talk just to hear yourself speak but I doubt you get out much.

closed 2021-04-12 by KevinOConnor