Klipper3d/klipper #4162: Problem with  Resonance Compensation

4noxx, 2021-04-11:
Klipper will not start if i add 

[mcu rpi]
serial = /tmp/klipper_host_mcu

[adxl345]
cs_pin = rpi:None

[resonance_tester]
accel_chip = adxl345
probe_points = 
	100,100,20

dont no why. without this, klipper starts

[klippy.log](https://github.com/KevinOConnor/klipper/files/6291756/klippy.log)

theopensourcerer, 2021-04-11:
"Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu'"

Have you followed the instructions for setting up the RPi as a mcu?

Also please read:

https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

4noxx, 2021-04-11:
> "Unable to open port: [Errno 2] No such file or directory: '/tmp/klipper_host_mcu'"
> 
> Have you followed the instructions for setting up the RPi as a mcu?
> 
> Also please read:
> 
> https://github.com/KevinOConnor/klipper/blob/master/docs/Contact.md#help-it-doesnt-work

i ve done this points here:
https://github.com/KevinOConnor/klipper/blob/master/docs/RPi_microcontroller.md

but did not help

4noxx, 2021-04-16:
need i a firmware with LINUX_PROCESS? like here https://www.youtube.com/watch?v=OoWQUcFimX8&t=391s
i ve a SKR 1.3 Board and can only flash with a sd card. 
if i make a firmware with LINUX PROCESS, if get no firmware.bin (klipper.bin) only a klipper.elf

closed 2021-04-17 by KevinOConnor