Klipper3d/klipper #4172: klipper "shutdown" when extruding with extruder connected to second mcu

jappyjan, 2021-04-15:
I am trying to controll additional stepper motors (extruders) through a second mcu.
The main mcu will control all the "normal" steppers (x, y, z, direct drive extruder) and the second one will controll 4 additional extruder motors.

The end goal is to have allways one of the additional steppers and the main direct drive extruder running in sync to allow for multi color printing.

the problem i ran into is that whenever i try to extrude with a stepper conntected to the second mcu klipper just shuts down.
it doesnt matter if i set the stepper up as the "normal" extruder or an extruder_stepper.

i also tried configuring one of the additional steppers as the "main" extruder and not configuring any additional extruders, this ended in the same result.

i am using klipper with moonraker on a pi 4.
my main mcu is a skr mini e3 v2.0 and the second one is a skr mini e3 v1.2

i attached the last config i tried which is setup to have the direct drive extruder connected to the main mcu and one of the steppers of the secondary mcu as a extruder_stepper.

[klippy.log](https://github.com/KevinOConnor/klipper/files/6320753/klippy.log)

jakep82, 2021-04-15:
You should request help for this on Discord or Discourse.  Klipper is unable to communicate with your driver.

```Transition to shutdown state: Unable to read tmc uart 'direct_drive' register IFCNT```

jappyjan, 2021-04-16:
closed thanks to the discord :D awesome community!

closed 2021-04-16 by jappyjan