Klipper3d/klipper #4173: Unable to get calculated homing_positive_dir from stepper in macro

wizhippo, 2021-04-16:
On line https://github.com/KevinOConnor/klipper/blob/d77928b17ba6b32189033b3d6decdb5bcc7c342c/klippy/stepper.py#L307 homing_positive_dir  is calculated but config is not updated to match so I cannot access the correct value from macros.

Example use would be

```
G1 X{'' if printer.toolhead.steppers[0].homing_positive_dir else '-'}{SAFE_HOME_DISTANCE}
```

This would allow  sensorless homing to back off in the correct direction using a generic macro

If supported a macro such as the following could be used

```
[homing_override]
axes: xyz
gcode:
  # Assumes both stepper x/y have same settings
  {% set driver_config = printer.configfile.settings['tmc2130 stepper_x'] %}
  {% set RUN_CUR = driver_config.run_current %}
  {% set HOLD_CUR = driver_config.hold_current %}
  {% set HOME_CUR = driver_config.run_current * 0.5 %}
  {% set SAFE_HOME_DISTANCE = 10 %}
  {% set X = params.X|default(false) %}
  {% set Y = params.Y|default(false) %}
  {% set Z = params.Z|default(false) %}
  {% set ALL = X is sameas false and Y is sameas false and Z is sameas false %}
  {% if X is not sameas false or ALL %}
  SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR} HOLDCURRENT={HOME_CUR * 0.5}
  G28 X
  SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CUR} HOLDCURRENT={HOLD_CUR}
  G91
  G1 X{'' if printer.toolhead.steppers[0].homing_positive_dir else '-'}{SAFE_HOME_DISTANCE}
  G90
  {% endif %}
  {% if Y is not sameas false or ALL %}
  SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR} HOLDCURRENT={HOME_CUR * 0.5}
  G28 Y
  SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CUR} HOLDCURRENT={HOLD_CUR}
  G91
  G1 Y{'' if printer.toolhead.steppers[1].homing_positive_dir else '-'}{SAFE_HOME_DISTANCE}
  G90
  {% endif %}
  {% if Z is not sameas false or ALL %}
  G28 Z0
  {% endif %}
```

wizhippo, 2021-04-16:
Found alternative, would still be nice to note have to duplicate in jinja

```
[homing_override]
axes: xyz
gcode:
  # Assumes both stepper x/y have same settings
  {% set driver_config = printer.configfile.settings['tmc2130 stepper_x'] %}
  {% set RUN_CUR = driver_config.run_current %}
  {% set HOLD_CUR = driver_config.hold_current %}
  {% set HOME_CUR = driver_config.run_current * 0.5 %}
  {% set SAFE_HOME_DISTANCE = 10 %}
  {% set X = params.X|default(false) %}
  {% set Y = params.Y|default(false) %}
  {% set Z = params.Z|default(false) %}
  {% set ALL = X is sameas false and Y is sameas false and Z is sameas false %}
  
  {% set homing_positive_dir_x = printer.configfile.settings.stepper_x.homing_positive_dir|default(none) %}
  {% set position_max_x = printer.configfile.settings.stepper_x.position_max|float %}
  {% set position_min_x = printer.configfile.settings.stepper_x.position_min|float %}
  {% set position_endstop_x = printer.configfile.settings.stepper_x.position_endstop|float %}
  {% if homing_positive_dir_x is none %}
    {% set axis_len_x = position_max_x - position_min_x %}
    {% if position_endstop_x <= position_min_x + axis_len_x / 4.0 %}
        {% set homing_positive_dir_x = false %}
    {% elif position_endstop_x >= position_max_x - axis_len_x / 4.0 %}
        {% set homing_positive_dir_x = true %}
    {% endif %}
  {% endif %}
  {% if X is not sameas false or ALL %}
  SET_TMC_CURRENT STEPPER=stepper_x CURRENT={HOME_CUR} HOLDCURRENT={HOME_CUR * 0.5}
  G28 X
  SET_TMC_CURRENT STEPPER=stepper_x CURRENT={RUN_CUR} HOLDCURRENT={HOLD_CUR}
  G91
  RESPOND MSG="G1 X{'-' if homing_positive_dir_x else ''}{SAFE_HOME_DISTANCE}"
  G90
  {% endif %}

  {% set homing_positive_dir_y = printer.configfile.settings.stepper_y.homing_positive_dir|default(none) %}
  {% set position_max_y = printer.configfile.settings.stepper_y.position_max|float %}
  {% set position_min_y = printer.configfile.settings.stepper_y.position_min|float %}
  {% set position_endstop_y = printer.configfile.settings.stepper_y.position_endstop|float %}
  {% if homing_positive_dir_y is none %}
    {% set axis_len_y = position_max_y - position_min_y %}
    {% if position_endstop_y <= position_min_y + axis_len_y / 4.0 %}
        {% set homing_positive_dir_y = false %}
    {% elif position_endstop_y >= position_max_y - axis_len_y / 4.0 %}
        {% set homing_positive_dir_y = true %}
    {% endif %}
  {% endif %}
  {% if Y is not sameas false or ALL %}
  SET_TMC_CURRENT STEPPER=stepper_y CURRENT={HOME_CUR} HOLDCURRENT={HOME_CUR * 0.5}
  G28 Y
  SET_TMC_CURRENT STEPPER=stepper_y CURRENT={RUN_CUR} HOLDCURRENT={HOLD_CUR}
  G91
  RESPOND MSG="G1 Y{'-' if homing_positive_dir_y else ''}{SAFE_HOME_DISTANCE}"
  G90
  {% endif %}

  {% if Z is not sameas false or ALL %}
  G28 Z0
  {% endif %}
```

closed 2021-04-16 by wizhippo

KevinOConnor, 2021-04-16:
FYI, I added this to the code (commit ca1d4316).  You should now be able to obtain the calculated default using `{printer.configfile.settings.stepper_x.homing_positive_dir}`.

-Kevin

wizhippo, 2021-04-16:
> FYI, I added this to the code (commit [ca1d431](https://github.com/KevinOConnor/klipper/commit/ca1d4316536bff1bc91ab08dfd5aba6c0575af86)). You should now be able to obtain the calculated default using `{printer.configfile.settings.stepper_x.homing_positive_dir}`.
> 
> -Kevin

Shout instead of 'getboolean' should it not be 'setboolean'?

Otherwise that will be great thanks