Klipper3d/klipper #4179: Error loading kinematics 'cartesian'

kornflex, 2021-04-17:
Hello,

Everythong works fine but with latest update, I have this error: Error loading kinematics 'cartesian'

I have updated OS, klipper 0.9.1-427, moonraker 0.5.0-18, fluidd 1.12.2

In klippy.log I have :

Error loading kinematics 'cartesian'
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/toolhead.py", line 259, in __init__
    self.kin = mod.load_kinematics(self, config)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 149, in load_kinematics
    return CartKinematics(toolhead, config)
  File "/home/pi/klipper/klippy/kinematics/cartesian.py", line 16, in __init__
    for n in 'xyz']
AttributeError: 'module' object has no attribute 'LookupMultiRail'
Config error
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 146, in _connect
    self._read_config()
  File "/home/pi/klipper/klippy/klippy.py", line 141, in _read_config
    m.add_printer_objects(config)
  File "/home/pi/klipper/klippy/toolhead.py", line 599, in add_printer_objects
    config.get_printer().add_object('toolhead', ToolHead(config))
  File "/home/pi/klipper/klippy/toolhead.py", line 267, in __init__
    raise config.error(msg)
Error: Error loading kinematics 'cartesian'
webhooks client 1969111808: New connection
webhooks client 1969111808: Client info {'program': 'Moonraker', 'version': 'v0.5.0-18-ge8f6862'}

I think there is a problem with python

Do you have an advice ?

Thank you

KevinOConnor, 2021-04-17:
Sorry - we don't use github to provide assistance.  That said, these types of errors are usually due to a corrupt sdcard, corrupt github tree, or corrupt OS image.

-Kevin

kornflex, 2021-04-18:
No reply help my on discord or facebook group, that's why I post here :/

SDcard is OK. I had to checkout the commit of 15 april to make klipper work.
If I update to the latest commit I have the error.

closed 2021-04-20 by github-actions[bot]