Klipper3d/klipper #4182: SKR 2 Unable to read tmc uart 'stepper_x' register IFCNT

MarceloFS13, 2021-04-17:
Hello!

I getting this error on the new SKR 2 board, i think this issue is related tho the stepper drivers but im using the same ones that i was using before i changed the board and they were working fine

[klippy (5).log](https://github.com/KevinOConnor/klipper/files/6330110/klippy.5.log)

MarceloFS13, 2021-04-18:
Its likely that we need to set a pin to boot with the MCU

closed 2021-04-20 by github-actions[bot]

VinnyCordeiro, 2021-04-20:
> 
> 
> Its likely that we need to set a pin to boot with the MCU.
> Marlin is having the same issue

Bigtreetech released the board's schematics [here](https://github.com/bigtreetech/SKR-2/blob/master/Hardware/BIGTREETECH%20SKR%202-SCH.pdf). Unlike other boards, on this one there is no MOSFET controlling the USB port, so as far as I understand this schematics, there is no need to set a port on compilation time.

I'm still studying the schematics to see if I'm missing something that can be related to this issue.

KevinOConnor, 2021-04-20:
It's probably the "mot power" pin.  If one adds the following to the config, does it fix it:
```
[output_pin mot_power]
value: 1
```

-Kevin

AndrezinhoCosta, 2021-06-26:
> It's probably the "mot power" pin. If one adds the following to the config, does it fix it:
> 
> ```
> [output_pin mot_power]
> pin: PC13
> value: 1
> ```
> 
> -Kevin

It seems to work well in SKR2. But I wanna know what pin I have to use with SKR Pro 1.2. I'm looking in GItHub near 2 days and can't find nothing!