Klipper3d/klipper #4185: Klipper + TMC2209 v3.0 (FYSETC) + SKR v1.3: Undervoltage GSTAT + Layershifts

JanKolibri, 2021-04-18:
Dear Community,

I can't get my SKR v1.3 + 4x TMC2209 running with Klipper and hope you could give me some help. Thank you in advance.

I switched from Marlin to Klipper recently. So please excuse my lack of depth. Please see the attachment of the klippy.log and a descriptive image.

The hardware in use, is the following:
Mainboard: BTT SKR v1.3 (Fig.: (B))
Stepper Driver: 4x TMC2209 v3.0 (Fig.: (C), (D))
Stepper Motor: 4x Nema17
Voltage Extruder: 1.0V
Voltage X,Y Stepper: 1.0V
Voltage Z Stepper: 0.76V
PSU: 24V

The following problems occurred.

Error: GSTAT: 00000004 uv_cp=1 (undervoltage) keeped stopping my prints (Fig.: (A)), eventhough I used this configuration on Marlin for months and Klipper didn't have issues before (maybe changes through update?). Considering UART might be the problem, I switched the configuration of Fig. (B) to [https://forum.fysetc.com/d/199-tmc2209-v3-0-uart/11](url), also by removing the SP-Pin (Fig.: (C), 8 and (D), 8) also. This made no difference, so I switched to the old configuration of Fig. (B).

According to #1952, I changed the microsteps of the stepper X, Y and the extruder from 32 to 16, with no more undervoltage. But now I get random layershifts for X and Y (Fig.: (E), 7). As a side by side comparison, I documented the change of the registry for the X driver in Fig. (F). I don't know how to interpret it and would appreciate your help. 

According to #2516 I tried the following with all stepper drivers, but no change:
SET_TMC_FIELD STEPPER=stepper_x FIELD=reset VALUE=1
SET_TMC_FIELD STEPPER=stepper_x FIELD=uvcp VALUE=1

I am now out of ideas and really would appreciate your input. 

Best regards,
Jan (ツ)╯

PS: I hope this post corresponds with the contribuition guidelines. If needed, changes are welcome.

[image]

[klippy.log](https://github.com/KevinOConnor/klipper/files/6331650/klippy.log)

KevinOConnor, 2021-04-18:
Sorry, we don't use github for support.  You should ask this question on Discord or Discourse.

-Kevin

JanKolibri, 2021-04-18:
Thank you, Kevin. I was not sure where to direct my question, because I find those directions rather confusing. Sorry for my mistake.

I did use Discord. Unfortuantely I am still stuck with this problem, because we were not able to solve it.
Could you please direct my to a different platform? Should I ask it at [https://klipper.discourse.group/](url) again?
Thanks in advance,
Jan (ツ)╯

JanKolibri, 2021-04-18:
I created the same topic again at: https://klipper.discourse.group/t/klipper-tmc2209-v3-0-fysetc-skr-v1-3-undervoltage-gstat-layershifts/151

Best regards,
Jan

closed 2021-04-20 by github-actions[bot]

JanKolibri, 2021-04-22:
@Kevin: Annoyingly my posts are being overly moderated at klipper.discource.group. Is this intended? I can't work like this.

I did some tinkering the last couples of days and fixed the issue for now. Here are my results.
I excluded problems with programing and mechanics, by comparing code and exchanging parts.
Recurring error message were (exp. Fig.: (A)):

    GSTAT: 00000004
    tmc uart register GCONF
    Internal error on command:“G1”.

Also in particular the extruder had troubles with starting point of the extrusion. As you can see in Fig. (D), 1 the steppes of the extrusion were missing, but only at the very beginning. After changing the layer and retracting the filament, gaps occured. Steps of the motor were calibrated, no clogging of the nozzle was present and the temperature of the motor was 55°C.

I did two things;
For the y-stepper I exchanged the faulty stepper drivers TMC2209 V3.0 (FYSETC) (Fig.: (E)) with a new one (Fig.: (B)). Accordingly to [2355](https://github.com/KevinOConnor/klipper/pull/2355) and [2243](https://github.com/KevinOConnor/klipper/pull/2243) the reseting of the GSTAT is intended.

As a second experiment, I exchanged the TMC2209 with a new TMC2208, because it was only available (Fig.: (C)). I noticed a change in of IOIN@TMC220X (Fig.: (C), 1). This could be related to the problems with the UART.

I am not sure if is due to the damaged chip are maybe a faulty component (i.e. resistor, capacitor). I will double check this in the future.

For now, this issue can be considered as solved.

Thank you for your help and stay healthy,
Jan (ツ)╯
[image]