Klipper3d/klipper #4193: Installation instructions re-write

LazaroFilm, 2021-04-20:
On the Klipper Facebook group, I have noticed that most of the issues new Klipper users have are related to the fact that they're using Octoprint, and after further discussion with them, it seems that that decision was influenced by the fact that the official documentation still recommends to install OctoPi in the "[installing](https://www.klipper3d.org/Installation.html)" link. 
Octoprint has some serious issues with klipper and is often the weak link when attempting to print fast or printing a high detailed object. 
Could we add a paragraph at the top explaining those issues, and offering the installation of Fluidd/Mainsail with Moonraker as an alternative? Maybe even suggesting FluiddPi and MainsailOS as simple install packages. 

Thanks for your help.

KevinOConnor, 2021-04-20:
Okay, but if this is a feature request note that we don't use github for requests.  If you'd like to discuss, maybe the Klipper Discourse is a better location.

-Kevin

KevinOConnor, 2021-05-06:
Looks like this got moved to https://klipper.discourse.group/t/installation-instructions-re-write-to-include-fluidd-mainsail/236/5 so I'm going to close this issue.

-Kevin

closed 2021-05-06 by KevinOConnor