Klipper3d/klipper #4199: Unexpected print head movements after update to commit acd949

cbc02009, 2021-04-22:
After updating klipper this morning to commit acd94909bc8b95c5f96f34a0f15e603b69b83246 from c5f1835becd88341202c1ec5fddd97009fe0491f (or close to that)  and now immediately after the homing routine my Voron 2.4 does very fast jerky movements to the side of the printbed and loses steps in the process causing it to crash into the y gantry.

Reverting back to commit c5f1835becd88341202c1ec5fddd97009fe0491f fixes the issue. Not sure what other helpful information I can provide, but please let me know. I've attached the klipper logs and the printer configuration.

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/6358468/printer.cfg.txt)
[klippy.log](https://github.com/KevinOConnor/klipper/files/6358469/klippy.log)

KevinOConnor, 2021-04-22:
That's unfortunate.  However, we don't use github to help diagnose problems with a printer (see docs/Contact.md ).  If you'd like to diagnose this issue you can use `git bisect` or similar to determine the commit that caused the regression.  Otherwise, best to use Klipper Discord or Klipper Discourse for assistance.

-Kevin

cbc02009, 2021-04-22:
> That's unfortunate. However, we don't use github to help diagnose problems with a printer (see docs/Contact.md ). If you'd like to diagnose this issue you can use `git bisect` or similar to [...]

I read through the contacts page before submitting the bug report. Because I was able to prove that undoing the update fixed the issue, I felt it was warranted to submit a bug report. Apologies if I misunderstood the directions.

 Seeing as there were no obvious changes to the repo that would cause this, I will do incremental upgrades and see if a particular commit can reproduce my issue. Do you want me to close this issue and start a new one if I find something more specific, or should I keep this one open pending results?

KevinOConnor, 2021-04-22:
Thanks.  It would be great if you can determine the particular commit and report it here.

-Kevin

cbc02009, 2021-04-23:
> 
> 
> Thanks. It would be great if you can determine the particular commit and report it here.
> 
> -Kevin

Of course I go to test this today and I can't reproduce the issue no matter what I do, even though I could yesterday. Must be some other variables I'm not taking into account. Sorry about this, I'll be sure to do better testing next time.

closed 2021-04-23 by cbc02009