Klipper3d/klipper #4217: !! Move out of range: After Klipper upgrade

xmthx, 2021-04-26:
Hi guys, i have upgrade Klipper at last version, and after the update literally i cant print because every prin that i try i see this error on the console: 

!! Move out of range: 197.200 375.000 12.560 [85.000]

My printer is CR10S Pro v2 (with original bltouch)

[printer.cfg.txt](https://github.com/KevinOConnor/klipper/files/6373802/printer.cfg.txt)

my gcode start from cura is this:
`M204 P500.00 R1000.00 T500.00 ;Setup Print/Retract/Travel acceleration
M205 X8.00 Y8.00 Z0.40 E5.00 ;Setup Jerk
M220 S100 ;Reset Feedrate
M221 S100 ;Reset Flowrate

G28 ;Home
;G29 ;BLTouch Leveling Bed
BED_MESH_CLEAR
BED_MESH_CALIBRATE

G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up`

Some one can help me? I'm not expert with klipper and i'm going crazy after the upgrade.

132print, 2021-04-26:
Hi, i had the same problems on my ender3pro.
The problem was the startcode from cura. So as first step i erased it.
(my startcode)
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up
G1 X10.1 Y20 Z0.28 F5000.0 ;Move to start position
G1 X10.1 Y200.0 Z0.28 F1500.0 E15 ;Draw the first line
G1 X10.4 Y200.0 Z0.28 F5000.0 ;Move to side a little
G1 X10.4 Y20 Z0.28 F1500.0 E30 ;Draw the second line
G92 E0 ;Reset Extruder
G1 Z2.0 F3000 ;Move Z Axis up`

As i have a lot to print at the moment, there's no time to really solve the problem.
For the beginning there's a way to do the startcode by klipper itself in configs.
have a try, good luck

xmthx, 2021-04-26:
problem solved, but i don't know how. Reboot many time and then work...!!!

closed 2021-04-28 by github-actions[bot]