Klipper3d/klipper #4219: Can't establish serial connection to MKS Robin E3D v1.1

piontec, 2021-04-26:
I've configured the board with STM32F103, 20KiB bootloader, 8MHz crystal and Serial (on USART1 PA10/PA9). After flashing, the serial port appears:

```
pi@octopi:~/klipper $ ls /dev/serial/by-id/*
/dev/serial/by-id/usb-1a86_USB_Serial-if00-port0
```

but I can't establish connection (see `terminal.log`). On exactly the same hardware setup, Marlin works.

[klippy.log](https://github.com/KevinOConnor/klipper/files/6375655/klippy.log)
[terminal.log](https://github.com/KevinOConnor/klipper/files/6375653/terminal.log)

piontec, 2021-04-26:
@KevinOConnor this is getting ridiculous: what I'm missing in this bug report that it is marked as "invalid"?

closed 2021-04-28 by github-actions[bot]

fex42, 2021-11-09:
I suffer from the same issue with my Robin Nano E3D board. Can't see why the ticket is closed... is there another issue for the problem?

With the tagged version v0.10.0 there is no problem, but sadly since some days git HEAD version of Klippy is not compatibly anymore and needs an MCU update (which is not possible with MKS Robin E3D because of this bug).