Klipper3d/klipper #4239: After the lateste update, i only get this message. I recompiled at least 100 times, deleted and cloned the git repo. I have now a huge paperweight...

anastazya, 2021-05-01:
Command format mismatch: adxl345_end oid=%c end1_clock=%u end2_clock=%u limit_count=%hu sequence=%hu vs adxl345_end oid=%c end1_time=%u end2_time=%u limit_count=%hu sequence=%hu
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

anastazya, 2021-05-01:
LE : ALL my 4 printers are now in this state after the update. I honestly don't know what to do. I followed this procedure : https://www.klipper3d.org/SDCard_Updates.html 

It doesn't work. Firmware is compiled and uploaded with success, but the printer throws the same error

gbeynon99, 2021-05-01:
You need to recompile and flash the firmare...this worked grat on my Voron, but I cannot now get my Ender 3 V2 to connect

b3sky, 2021-05-01:
I have a problem too.
after recompile and firmware reflash
robin nano v3 -> fine.
skr 1.4 turbo -> lost usb connection

Iony3d, 2021-05-01:
Command format mismatch: endstop_state oid=%c homing=%c next_clock=%u pin_value=%c vs endstop_state oid=%c homing=%c pin_value=%c
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

same problem after update

closed 2021-05-03 by github-actions[bot]