Klipper3d/klipper #4244: Ender-2 LCD on SKR Mini E3 v2.0

Asterchades, 2021-05-01:
As mentioned in the title, I'm attempting to use the stock LCD screen (similar to an MKS Mini 12864) from a Creality Ender-2 on an SKR Mini E3 v2.0 mainboard. Under Marlin this is as simple as enabling the "ENDER2_DISPLAY" flag when compiling the firmware, so I know from experience that the combination is compatible. Naturally I don't expect the same flag to exist in Klipper but I'm running across an issue with trying to replicate the Marlin config, so perhaps there's something I'm missing or something that needs fixing.

I have the display configured as follows:

`[board_pins]
aliases:
    # EXP1 header
		EXP1_1=<5V>, EXP1_3=PB15, EXP1_5=PB9, EXP1_7=<RST>, EXP1_9=PA15,
		EXP1_2=<GND>, EXP1_4=PB8, EXP1_6=PA10, EXP1_8=PA9, EXP1_10=PB5

[display]
lcd_type: uc1701
cs_pin: PB8
a0_pin: PB9
encoder_pins: ^PA9, ^PA10
click_pin: ^!PA15
#spi_software_miso_pin:
spi_software_mosi_pin: PB15
spi_software_sclk_pin: PB5`

First, you may have noticed that I've reversed the order of the pin aliases as compared to the example config for this controller. This replicates the layout found in Marlin and also matches the physical layout of the board (EXP1_1 is designated by the square solder pad, and located at the front left corner of the board when looking at the card slot/USB port) but ultimately is not relevant to the issue at hand (unless Klipper internally reverses the order of the pins). I include this information merely for completeness.

In the actual display config, there's a problem: there simply isn't enough pins for all of the requested settings. A 10-pin header, with three reserved pins (+5v, GND, RST) leaves only seven pins for eight values. The pin I have not been able to find an appropriate allocation for is the MISO pin, which I cannot find any reference to for this screen in Marlin at all. But if I leave out the value I get an error requesting it, if I assign it to any other valid pin name I get an error for a conflict, setting it to "null" shuts down the MCU, and if I remove the software SPI references entirely I get an error relating to SPI2 on my Z axis driver.

As best I can find, this screen uses software SPI and has no MISO facility. But I cannot find a way to get Klipper to accept this.

Asterchades, 2021-05-02:
Small update. I have found a work-around which, while it appears successful, I'm not at all happy with.

Specifically I have redirected spi_software_miso_pin to PA8 - the signal pin for the NeoPixel header. As I don't have a NeoPixel attached (or any intention of doing so) this is simply a throw-away pin that shouldn't interfere with anything. But it would obviously cause problems for anyone who does have a NeoPixel attempting to use the same (or a similar) display. With this in place, however, the screen works as it should. Or, rather, it did as soon as I reversed the encoder pins to correct the direction.

To that end, I feel that at least this screen type (or perhaps a variant name - uc1701c?) should not require that particular parameter.

KevinOConnor, 2021-05-02:
Sorry, but we don't use github to help configure individual printers.  If you're looking for feedback on a config, you can raise this on Discord or Discourse.

-Kevin

closed 2021-05-03 by Asterchades