Klipper3d/klipper #4245: Same issue someone else had with update

AMRaisch, 2021-05-01:
Command format mismatch: adxl345_end oid=%c end1_clock=%u end2_clock=%u limit_count=%hu sequence=%hu vs adxl345_end oid=%c end1_time=%u end2_time=%u limit_count=%hu sequence=%hu
This type of error is frequently caused by running an older
version of the firmware on the micro-controller (fix by
recompiling and flashing the firmware).
Once the underlying issue is corrected, use the "RESTART"
command to reload the config and restart the host software.
Protocol error connecting to printer

rather than having the bot suspend the post how about you fix the issue.  Before update printer worked.  After update printer did not!  Fix it!

AMRaisch, 2021-05-01:
By the way this is the second time I have updated the firmware and spent hours trying to solve a problem that you created!  I love klipper but this is a complete waste of my time and resources!

AMRaisch, 2021-05-01:
Do not label this invalid until the problem is solved!

kpochwala, 2021-05-01:
Dont use it then man, or use tagged branch.

Booner99, 2021-05-01:
I have the problem too.

kpochwala, 2021-05-01:
AFAIK there were some commits regarding ADXL errors support: https://github.com/KevinOConnor/klipper/pull/4196 you can check if this is the thing causing problems. 

Rather than going around throwing "hurr durr fix it because I say so" take a sip of tea, look around the pull requests or commit messages and try at least pointing to some direction @AMRaisch.

AMRaisch, 2021-05-01:
kpochwala  some of us do not have time to do a bunch of research on how to fix an error that occurs by simply pressing the link for an update.  I did not start using klipper so that I could learn how to trouble shoot klipper or program python.  I started using it for the end user experience and increased functionality!

AMRaisch, 2021-05-01:
I am also more concerned about my productivity than someone's feelings getting hurt!

AMRaisch, 2021-05-01:
It would also be a handy tool to be able to revert back to a previous installation just in case the new one does not work!

BlackStump, 2021-05-02:
you can't argue or reason with idiots they bring you down to their level then beat you with experience.

AMRaisch, 2021-05-02:
Pithy sayings offer no remediation!  Reasoning with me would be to offer a logical solution to the problem that, not only I am having but, apparently many others as well!

BlackStump, 2021-05-02:
I have updated my firmware without issue!

AMRaisch, 2021-05-02:
Good for you!  And a group of us did not!  Now would you like to offer something of content?  Or do you just want to continue trying to assure me of your intelligence?

kpochwala, 2021-05-02:
> It would also be a handy tool to be able to revert back to a previous installation just in case the new one does not work!

There is. It's called git: 
`git reflog # to see what have you done`
`git checkout <where do you want to be>`

If you updated using some third party UI, such as fluidd or mainsail, then please ask respective owners for support to revert to the previous version using GUI. This repository is for klipper, not fluidd/mainsail nor even moonraker.

> @kpochwala some of us do not have time to do a bunch of research on how to fix an error that occurs by simply pressing the link for an update. I did not start using klipper so that I could learn how [...]

Then as I said earlier, use tagged version, masters are *virtually always* buggy, no matter what open source program sources you use.

BlackStump, 2021-05-02:
The point is you have given no info at all apart from a error message which describes the solution.
It doesn't work - not helpful
what exactly have you done apart from complain!
what hardware are you having issues with?
Complaining and doing nothing is BS follow the directions before opening a issue.
As I see it you have a choice use the software that has been provided or don't use that is your choice.
It is not your choice to make demands!
Show some respect to the developer!

AMRaisch, 2021-05-02:
@kpochwala Thank you!  This is very helpful information!  I had no idea that the updates offered through the GUI were not "tagged"  I was not even aware of the difference between "tagged" and "master"  I know "alpha" "beta" and "official" as standard designators of developmental status for software.  Again I am an end user.

AMRaisch, 2021-05-02:
@BlackStump I did not realize this was a hardware issue.  As I stated in my initial post, the printer was running Klipper prior to the upgrade.  So I stated the only thing that changed.  Which was the upgraded firmware.  I also do not care about arbitrarily/subjectively determined directions!  I know that I have an issue.  I know the issue stemmed from the update.  I know that the source of the update is on this git repository.  

I pulled the git 3 times.  I reflashed the sd card 4 times.  I opened the python scripts in my ssh to see if i could find the mismatch.  I joined the discord to see if I might find a section concerning update/installation issues.  These are all things that I did not want to do nor did I plan to do when I clicked a button that said an update was available. 

I am trying to use the software.  That is specifically the problem.  If I were trying to use a new feature or had just installed a new piece of hardware that I needed to configure, I would assume some headache.  But I did not.  I just clicked the update button!

The hardware is a Troodon 300 running klipper via pi 4.  

There is a big difference between the expectations of the people who contribute to this git and the people who simply want to spend time calibrating there printer/slicer settings to optimize print time and quality.  Also seeing how Duet, Marlin, and Klipper are all freeware, the options are pretty limited when it comes to choosing a payed firmware.  Does it stand to reason that no home 3D printer user should have expectations of the developers?  When I download the latest release of Blender or FreeCAD, I don't have issues.  But if I were in the middle of trying to design something for a client and an update locked me out of my computer, I think it would be pretty reasonable of me to be upset.  Or should I have chosen not to use the software or be so familiar with it that I could fix the code?   This is simply not in my wheelhouse.  Nor do I, at this point in time, wish it to be.  I just want a reliable/repeatable 3D printing experience!

zellneralex, 2021-05-02:
Btw here is your solution: https://www.klipper3d.org/RPi_microcontroller.html#building-the-micro-controller-code. You have an adxl and a mcu on the pi defined therefor you need also to rebuild that mcu code

Arksine, 2021-05-02:
> @kpochwala Thank you! This is very helpful information! I had no idea that the updates offered through the GUI were not "tagged" I was not even aware of the difference between "tagged" and "master" [...]

Moonraker is tagged as alpha, there hasn't been an official release yet.   It is intended for users willing to deal with issues encountered through dev channel updates.   The ultimate goal is to provide end users with the ease of use and stability, but we are not there yet.  That is why Moonraker's `update_manger` is updating the dev branch.

If stability is what you want my suggestion is to use Octoprint and and stick to official releases of Klipper.  If you want the latest and greatest then we welcome you to use Moonraker and one of its clients, just know that at times you may have to deal with breakage that requires some attention on your part.

kpochwala, 2021-05-02:
> @BlackStump I did not realize this was a hardware issue. As I stated in my initial post, the printer was running Klipper prior to the upgrade. So I stated the only thing that changed. Which was the [...]

Have you ever used master branch of Blender? :) Go for it (it is currently 3.0 development), you will see how the software is created. Builds that you get from the website are again, tagged versions (and also alpha/beta - all features set in, only dealing with bugfixes).

> > @kpochwala Thank you! This is very helpful information! I had no idea that the updates offered through the GUI were not "tagged" I was not even aware of the difference between "tagged" and "master" I know "alpha" "beta" and "official" as standard designators of developmental status for software. Again I am an end user.
> 
> Moonraker is tagged as beta, there hasn't been an official release yet. It is intended for users willing to deal with issues encountered through dev channel updates. The ultimate goal is to provide end users with the ease of use and stability, but we are not there yet. That is why Moonraker's `update_manager` is updating the dev branch.
> 
> If stability is what you want my suggestion is to use Octoprint and and stick to official releases of Klipper. If you want the latest and greatest then we welcome you to use Moonraker and one of its clients, just know that at times you may have to deal with breakage that requires some attention on your part. In this case, you need to update all of your attached MCUs, as mentioned above.

Hmm, I'll look into `update_manager` class in Moonraker project, maybe adding some switch in `update_manager.conf` to fetch klipper from tags instead of latest commit would be viable? But it's a story for Moonraker issue :)

lowtolerance, 2021-05-02:
Jesus, the sense of entitlement from some of you is unreal. This is free, open source software and Kevin doesn’t owe you a goddamned thing.

AMRaisch, 2021-05-02:
@zellneralex @Arksine @kpochwala. Thank you all very much for the information.  I apologize for the attitude!  I will take this under advisement.

closed 2021-05-02 by AMRaisch