Klipper3d/klipper #4247: MKS Robin Mini unreachable over usb, maybe SDIO issue?

Fissh02, 2021-05-02:
[klippy (12).log](https://github.com/KevinOConnor/klipper/files/6411249/klippy.12.log)

[image]

Steps to reproduce: 
- follow the docs about flashing over sd card, using menuconfig setting that work on the same board for others:

Klipper Firmware Configuration
[*] Enable extra low-level configuration options
    Micro-controller Architecture (STMicroelectronics STM32)  --->
    Processor model (STM32F103)  --->
    Bootloader offset (28KiB bootloader)  --->
    Clock Reference (8 MHz crystal)  --->
[ ] Use USB for communication (instead of serial)
[ ] Use CAN for communication (instead of serial)
    Serial Port (USART3)  --->
(250000) Baud rate for serial port
[ ] Specify a custom step pulse duration
()  GPIO pins to set at micro-controller startup

- connect over usb (have tried use USB for communication (instead of serial)
- Upload correct printer.cfg 

“Unable to connect”

I read through the Sd card updates document, it mentioned being broken on boards that use SDIO. Could this be an issue if SDIO?

I’ve spoken to a few klipper gurus who’ve used this board and they thought it must be a bug. I have managed to run klipper on precompiled .bin files from others’ work, but they are too old to run on the newest version of klipper now. I am running on a raspberry pi zero w with mainsailos.

lowtolerance, 2021-05-02:
From your log it does not seem as though your firmware flash was successful. Try changing the file name from something other than klipper.bin and flashing it again.

Fissh02, 2021-05-02:
> From your log it does not seem as though your firmware flash was successful. Try changing the file name from something other than klipper.bin and flashing it again.

The flash was successful. The file was changed by the printer and the screen changed to show an update to the firmware.

lowtolerance, 2021-05-02:
Your settings look right to me, so if you’re sure the right bootloader has been flashed, forget I said that. 

Does the path to the serial device under the [mcu] heading of your printer.cfg actually exist? If listing the contents of /dev/ doesn’t show a directory named “serial” directory, then your firmware flash was probably unsuccessful. If it does, make sure the file names match.

KevinOConnor, 2021-05-02:
Sorry, but we don't use github to help diagnose problems with a printer.  Feel free to raise this on the Klipper Discord or Discourse servers.

-Kevin

closed 2021-05-04 by github-actions[bot]