Klipper3d/klipper #4249: G28 fails when when z_pos is near to z_max + z_hop is set & M84 is issued (or "idle timeout" has passed)

kmarty, 2021-05-02:
Hi,
I think there's incorrect behaviour of G28( Z) when last known_position and "_[safe_z_home] z_hop_" is higher than _position_max_ of Z axis.

Lets say I have z_position_max = 215 and z_hop = 10:
- execute homing G28
- move Z axis to the z_position_max (either by printing or just by G0 Z215)
- execute G28 again.

In this scenario, G28 correctly says "_// No zhop performed, target Z out of bounds: 225.0_", it _does_ the homing of axis and everything is fine.

But if:
- execute homing G28
- move Z axis to the z_position_max
- wait until idle timeout or issue M18/M84 without waiting
- execute G28

In this scenario homing fails with "_!! Move out of range: 137.500 141.400 225.000 [0.000]_" and no homing is performed. I have to execute G28 again to make real homing (with message "_// No zhop performed..._").

[klippy.log](https://github.com/KevinOConnor/klipper/files/6411783/klippy.log)

KevinOConnor, 2021-05-03:
@FHeilmann - do you have any thoughts on this?

-Kevin

FHeilmann, 2021-05-03:
@KevinOConnor this is caused by the fact that in his described scenario, `z` will not be among the homed axes, but klipper still keeps the last internal position of the z axis when the idle timeout occurs. `safe_z_home` will attempt to hop the z axis, because it is not homed, however the zhop, together with the last known position of z, will cause the move to go out of bounds.

This can probably be fixed by augmenting the if-statement here https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/safe_z_home.py#L40 to also skip the zhop when the last known z position would cause the zhop to go out of bounds.

kmarty, 2021-05-05:
> I'm not sure why it works if G28 is executed a 2nd time, and would have to investigate further.
I did some (primitive) "debug" messages using:
```diff
diff --git a/klippy/extras/safe_z_home.py b/klippy/extras/safe_z_home.py
index 6dd2f292..b9110ed0 100644
--- a/klippy/extras/safe_z_home.py
+++ b/klippy/extras/safe_z_home.py
@@ -3,6 +3,7 @@
 # Copyright (C) 2019 Florian Heilmann <Florian.Heilmann@gmx.net>
 #
 # This file may be distributed under the terms of the GNU GPLv3 license.
+import logging
 
 class SafeZHoming:
     def __init__(self, config):
@@ -30,6 +31,7 @@ class SafeZHoming:
                                +" be used simultaneously")
 
     def cmd_G28(self, gcmd):
+        logging.info("cmd_G28 - BEGIN")
         toolhead = self.printer.lookup_object('toolhead')
 
         # Perform Z Hop if necessary
@@ -37,6 +39,7 @@ class SafeZHoming:
             # Check if Z axis is homed or has a known position
             curtime = self.printer.get_reactor().monotonic()
             kin_status = toolhead.get_kinematics().get_status(curtime)
+            logging.info("kin_status['homed_axes'] = %s" % kin_status['homed_axes'])
             if 'z' in kin_status['homed_axes']:
                 # Check if the zhop would exceed the printer limits
                 pos = toolhead.get_position()
@@ -87,6 +90,7 @@ class SafeZHoming:
             # Move XY back to previous positions
             if self.move_to_previous:
                 toolhead.manual_move(prevpos[:2], self.speed)
+        logging.info("cmd_G28 - END")
 
     def _perform_z_hop(self):
         toolhead = self.printer.lookup_object('toolhead')
```
and first (unsuccessfull) G28 is:
```
cmd_G28 - BEGIN
kin_status['homed_axes'] = 
Move out of range: 137.500 141.400 225.000 [0.000]
```
while second (immediatelly after) G28:
```
cmd_G28 - BEGIN
kin_status['homed_axes'] = z
No zhop performed, target Z out of bounds: 225.0
cmd_G28 - END
```

To me, as non-developer, it sounds like the problem is quite deeper, it got Z-axis as homed without homing. Am I wrong?

kmarty, 2021-05-05:
> I'm not sure why it works if G28 is executed a 2nd time, and would have to investigate further.

Maybe I'm totally wrong, but it seems that https://github.com/KevinOConnor/klipper/blob/master/klippy/extras/safe_z_home.py#L95 sets "z" in "homed_axes" before Z is really homed.

I tried some (primitive) debug with:
```diff
diff --git a/klippy/extras/safe_z_home.py b/klippy/extras/safe_z_home.py
index 6dd2f292..5f17dbc4 100644
--- a/klippy/extras/safe_z_home.py
+++ b/klippy/extras/safe_z_home.py
@@ -3,6 +3,7 @@
 # Copyright (C) 2019 Florian Heilmann <Florian.Heilmann@gmx.net>
 #
 # This file may be distributed under the terms of the GNU GPLv3 license.
+import logging
 
 class SafeZHoming:
     def __init__(self, config):
@@ -30,6 +31,7 @@ class SafeZHoming:
                                +" be used simultaneously")
 
     def cmd_G28(self, gcmd):
+        logging.info("cmd_G28 - BEGIN")
         toolhead = self.printer.lookup_object('toolhead')
 
         # Perform Z Hop if necessary
@@ -87,13 +89,20 @@ class SafeZHoming:
             # Move XY back to previous positions
             if self.move_to_previous:
                 toolhead.manual_move(prevpos[:2], self.speed)
+        logging.info("cmd_G28 - END")
 
     def _perform_z_hop(self):
+        logging.info("_perform_z_hop - BEGIN")
         toolhead = self.printer.lookup_object('toolhead')
         # Perform the Z-Hop
         pos = toolhead.get_position()
+        kin_status_debug = toolhead.get_kinematics().get_status(self.printer.get_reactor().monotonic())
+        logging.info("Before: homed_axes = %s" % kin_status_debug['homed_axes'])
         toolhead.set_position(pos, homing_axes=[2])
+        kin_status_debug = toolhead.get_kinematics().get_status(self.printer.get_reactor().monotonic())
+        logging.info("After: homed_axes = %s" % kin_status_debug['homed_axes'])
         toolhead.manual_move([None, None, pos[2]+self.z_hop], self.z_hop_speed)
+        logging.info("_perform_z_hop - END")
 
 def load_config(config):
     return SafeZHoming(config)
```

And first (failed) G28 on this scenario returns:
```
cmd_G28 - BEGIN
_perform_z_hop - BEGIN
Before: homed_axes = 
After: homed_axes = z
Move out of range: 137.500 141.400 225.000 [0.000]
```
(no axis moves happened and "homed_axes" was set to "z")

While the second G28 (with "z" in "homed_axes" already set by previous, unsuccessful, attempt):
```
cmd_G28 - BEGIN
No zhop performed, target Z out of bounds: 225.0
cmd_G28 - END
```

Hope it somehow helps.

FHeilmann, 2021-05-06:
Let's ignore the "it works on 2nd attempt" problem for now, and let's try fixing the actual problem (the first instance failing).

can you try this changeset and see if that fixes the problem?

```diff
diff --git a/klippy/extras/safe_z_home.py b/klippy/extras/safe_z_home.py
index 6dd2f292..edfa411e 100644
--- a/klippy/extras/safe_z_home.py
+++ b/klippy/extras/safe_z_home.py
@@ -37,19 +37,19 @@ class SafeZHoming:
             # Check if Z axis is homed or has a known position
             curtime = self.printer.get_reactor().monotonic()
             kin_status = toolhead.get_kinematics().get_status(curtime)
-            if 'z' in kin_status['homed_axes']:
-                # Check if the zhop would exceed the printer limits
-                pos = toolhead.get_position()
-                if pos[2] + self.z_hop > self.max_z:
-                    gcmd.respond_info(
-                        "No zhop performed, target Z out of bounds: " +
-                        str(pos[2] + self.z_hop))
-                elif pos[2] < self.z_hop:
+            # Check if the zhop would exceed the printer limits
+            pos = toolhead.get_position()
+            if pos[2] + self.z_hop > self.max_z:
+                gcmd.respond_info(
+                    "No zhop performed, target Z out of bounds: " +
+                    str(pos[2] + self.z_hop))
+            elif pos[2] < self.z_hop:
+                if 'z' in kin_status['homed_axes']:
                     self._perform_z_hop()
-            else:
-                self._perform_z_hop()
-                if hasattr(toolhead.get_kinematics(), "note_z_not_homed"):
-                    toolhead.get_kinematics().note_z_not_homed()
+                else:
+                    self._perform_z_hop()
+                    if hasattr(toolhead.get_kinematics(), "note_z_not_homed"):
+                        toolhead.get_kinematics().note_z_not_homed()

         # Determine which axes we need to home
         need_x, need_y, need_z = [gcmd.get(axis, None) is not None
```

as mentioned earlier in my comment, this inverts the nesting of if-statements. The reason the initial attempts fail is that klipper still keeps the z position, even if the axis is not homed. The fact that the axis is not homed bypasses the limit check, which leads to the error. My proposed changeset performs the limit check always. So regardless of whether the axis is homed or not, if the last position makes klipper think it might go out of bounds, the zhop is skipped.

kmarty, 2021-05-06:
> can you try this changeset and see if that fixes the problem?

Seems it works fine.

KevinOConnor, 2021-05-26:
This should be resolved with the merge of #4271 .

-Kevin

closed 2021-06-03 by github-actions[bot]